jekhokie/raspberry-noaa-v2 #486: [Error] some packages are departed.. dead , uninstallable.

LeFEAR, 2022-01-02:
**Describe the bug**
A clear and concise description of what the bug is.
This is one of the many errors I tried to fix buy adding new versions or removing tat package all together. 
The platform is a Debian VM and I tried it also on a raspberry pi with the same issues. 

also these 2 packages are also uninstallable : 
Pillow==7.1.2
pylab==0.0.1
and 
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Dependency is not satisfiable: libjpeg-turbo8\n"}

BUT the other friend of mine that had issues was troubled with libusb .. 

also I used the pull request of this person in order to continue from where I was stuck : 
rhaamo:feature/non-pi-install

**Screenshots and/or logs**
If applicable, add screenshots to help explain your problem. Logs are always welcome as well.

non to show at this point. 

**Support Output**
Paste the output of running the `./support.sh` script here.

**Server information (please complete the following information):**
 - OS: Debian 11.2.0
 - Python version 2.7 and 3.9 
 - SDR device : adalm pluto ( haven't went that far yet ) 

**Additional context**
Add any other context about the problem here.
Thank you in advance for your time. /

It would be awesome to be able to run on a VM with Debian on an industrial scale server. 

Thanks

dom-robinson, 2022-01-03:
Feel free to port it! Currently it was working fine with Buster and is somewhere between Buster and Bullseye support.

LeFEAR, 2022-01-03:
> Feel free to port it! Currently it was working fine with Buster and is somewhere between Buster and Bullseye support.

Well the fork was by accident. :D 

never the less I will try those versions because I am not as proficient as you may think ... but I try to stay up to date with most stuff hence forth the latest debian image, 

never the less thank you for the help. I hope I wasn't a Nuisance.

mihajlo2003petkovic, 2023-10-04:
Hi! I've implemented native support for amd64 CPUs used by regular PCs, and now it works on Debian Bullseye based distros, I specifically tested ir on LMDE 5 (current LMDE 6 which came out a few days after I conducted my tests is Bookworm so use Bullseye). I'll close this for now, but if it still persists, I'll reopen the issue.

closed 2023-10-04 by mihajlo2003petkovic