jekhokie/raspberry-noaa-v2 #487: [Error] "AnsibleUndefinedVariable: 'noaa_receiver' is undefined"

TitleOS, 2022-01-02:
**Describe the bug**
A clear and concise description of what the bug is.

**Screenshots and/or logs**

> PLAY [install raspberry-noaa-v2 core] ******************************************
> 
> TASK [get os release] **********************************************************
> ok: [localhost]
> 
> TASK [common : noaa environment variables] *************************************
> fatal: [localhost]: FAILED! => {"changed": false, "msg": "AnsibleUndefinedVariable: 'noaa_receiver' is undefined"}
>         to retry, use: --limit @/home/pi/raspberry-noaa-v2/ansible/site.retry
> 
> PLAY RECAP *********************************************************************
> localhost                  : ok=1    changed=0    unreachable=0    failed=1
> 
> Error:   Something failed with the install - please inspect the logs above

**Server information (please complete the following information):**
 - OS: Raspbian 
 - Raspberry Pi Model: Raspberry Pi 3B+
 - Python version: 3.7.3
 - SDR device: Nooelec NESDR Smartee 

**Additional context**
When attempting to update my raspberry-noaa-v2 installation to the latest git commit of the master branch (By following the upgrade instructions on the repo's home page, running git pull then install_and_upgrade.sh), after executing install_and_upgrade.sh, I shortly run into the error mentioned above about the noaa_receiver variable being undefined.

dom-robinson, 2022-01-03:
Something has 'gone wrong' between updates to bullseye from 3.18 which was fully tested on Buster - if you can install buster and use tagged rel 1.8 that should work.

ve3elb, 2022-01-03:
Done the issue is with php7.4. Dee my post in the discord group I tagged you to it.

colinluthier, 2022-01-28:
Adding:

noaa_receiver: 'gnuradio'

to your settings.yml fixes this. I pushed a fix for all the new variables and they should be ready now.

closed 2022-01-28 by colinluthier