jekhokie/raspberry-noaa-v2 #499: [Error]

SeanMacSuibhne, 2022-01-17:
**Describe the bug**
Hi Been using RPi NOAA-V2 for months and decided to reinstall as I saw images from other peoples images were different from mine 
I moved ~/raspberry-noaa-v2 to ~/raspberry-noaa-v2_old
Recloned the repository
 cd to ~/raspberry-noaa-v2
and ran ./install_and_upgrade.sh 
that gave errors and I reran with the suggested options
./install_and_upgrade.sh  --limit @/home/pi/raspberry-noaa-v2/ansible/site.retry
but got the same errors

When I try to install I get this error
TASK [webserver : install php] *************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'php7.4-fpm' is available"}
	to retry, use: --limit @/home/pi/raspberry-noaa-v2/ansible/site.retry

PLAY RECAP *********************************************************************
localhost                  : ok=44   changed=0    unreachable=0    failed=1   

Error:   Something failed with the install - please inspect the logs above

**Screenshots and/or logs**
If applicable, add screenshots to help explain your problem. Logs are always welcome as well.

**Support Output**
Paste the output of running the `./support.sh` script here.
pi@P6:~/raspberry-noaa-v2 $ ./support.sh
=============================================
Details about environment
=============================================
Current local date/time: Mon Jan 17 18:22:26 GMT 2022
Current date/time (ms):  1642443746
Repo git hash:           30a07ff086ccd978e2dc1f962b3cc20330026687
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.2
Architecture:            Linux P6 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [4320	Mon Jan 17 19:59:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20220117-195943 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1642449583 968 39 Northbound W
  * [4314	Mon Jan 17 22:04:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20220117-220409 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1642457049 948 69 Northbound W
  * [4313	Mon Jan 17 20:24:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20220117-202442 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1642451082 887 24 Northbound E
  * [4309	Mon Jan 17 18:57:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20220117-185714 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1642445834 920 68 Northbound E
  * [4315	Mon Jan 17 23:47:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20220117-234730 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1642463250 770 12 Northbound W
---------------------------------------------
Satellite SDR Settings:
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: 0
  noaa_15_enable_bias_tee: true
  noaa_15_gain: 30
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 30
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: 0
  noaa_18_enable_bias_tee: true
  noaa_18_gain: 30
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 30
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: 0
  noaa_19_enable_bias_tee: true
  noaa_19_gain: 30
  noaa_19_sun_min_elevation: 6
  noaa_19_sat_min_elevation: 30
  meteor_m2_schedule: true
  meteor_m2_sdr_device_id: 0
  meteor_m2_freq_offset: 0
  meteor_m2_enable_bias_tee: true
  meteor_m2_gain: 30
  meteor_m2_sun_min_elevation: 6
  meteor_m2_sat_min_elevation: 30
  meteor_m2_memory_threshold: 600
---------------------------------------------
USB Device Map:
  Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
  Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
  Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
  mmcblk0     179:0    0   29G  0 disk 
  ├─mmcblk0p1 179:1    0  256M  0 part /boot
  └─mmcblk0p2 179:2    0 28.7G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks     Used Available Use% Mounted on
  /dev/root       29579428 23203012   5123564  82% /
  devtmpfs         1800564        0   1800564   0% /dev
  tmpfs            1965428    22708   1942720   2% /dev/shm
  tmpfs            1965428     8828   1956600   1% /run
  tmpfs               5120        4      5116   1% /run/lock
  tmpfs            1965428        0   1965428   0% /sys/fs/cgroup
  tmpfs            1024000        0   1024000   0% /var/ramfs
  /dev/mmcblk0p1    258095    49282    208813  20% /boot
  tmpfs             393084        8    393076   1% /run/user/1000
---------------------------------------------
Memory Info:
  MemTotal:        3930860 kB
  MemFree:          625720 kB
  MemAvailable:    3169476 kB
---------------------------------------------
Git source files changed:
  (None)
---------------------------------------------
SDR Information:
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer,
    foreign key(pass_start) references passes(pass_start));

You have new mail in /var/mail/pi

**Server information (please complete the following information):**
 - OS: [e.g. Raspbian]
 - Python version [e.g. 3.7.3]
 - SDR device [e.g. RTL-SDR v3]

pi@P6:~/raspberry-noaa-v2 $ python
Python 2.7.16 (default, Oct 10 2019, 22:02:15) 
[GCC 8.3.0] on linux2

pi@P6:~/raspberry-noaa-v2 $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"

**Additional context**
Add any other context about the problem here.

I did sudo apt update && sudo atp upgrade before installation.

Thank you
Seán

dom-robinson, 2022-01-17:
Hey Sean - yeah we are having a 'tween' moment between buster and bullseye. Hoping to have it resolved in a few days - if you go for tagged version 1.8 on buster that was known good in the meanwhile..

SeanMacSuibhne, 2022-01-18:
Thanks for the response and suggestion. I have installed that one now but I get a 403 error when I go to the localhost webpage. Maybe its because I have no new captured images yet?
Seán

colinluthier, 2022-01-28:
@SeanMacSuibhne Just making sure this worked out?

Cadair, 2022-01-28:
This should now be fixed in master.

closed 2022-01-28 by Cadair