jekhokie/raspberry-noaa-v2 #507: Wrong version of numpy

mattues, 2022-01-27:
Hi
I can´t get the polarization to show up at the image webpage. Noticed in the logs that after a 'sudo upgrade' my numpy version is wrong.
How can i fix this without breaking anything?

Pasting the logs for information

----------------------------------------------------------------------------------------------------------------------------------------------------

27-01-2022 19:39 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Producing polar graph of direction for pass
RuntimeError: module compiled against API version 0xe but this version of numpy is 0xd
Traceback (most recent call last):
  File "/home/pi/raspberry-noaa-v2/scripts/image_processors/polar_plot.py", line 31, in <module>
    import matplotlib.pyplot as plt
  File "/usr/local/lib/python3.9/dist-packages/matplotlib/__init__.py", line 174, in <module>
    _check_versions()
  File "/usr/local/lib/python3.9/dist-packages/matplotlib/__init__.py", line 159, in _check_versions
    from . import ft2font
ImportError: numpy.core.multiarray failed to import
convert: unable to open image `/srv/images/NOAA-18-20220127-192514-polar-direction.png': Filen eller katalogen finns inte @ error/blob.c/OpenBlob/2924.
convert: no images defined `/srv/images/thumb/NOAA-18-20220127-192514-polar-direction.png' @ error/convert.c/ConvertImageCommand/3229.

----------------------------------------------------------------------------------------------------------------------------------------------------

pi@raspberrypi:~/raspberry-noaa-v2 $ ./support.sh
=============================================
Details about environment
=============================================
Current local date/time: tor 27 jan 2022 20:43:02 CET
Current date/time (ms):  1643312582
Repo git hash:           2d8927b2f040c739c2f009b566e34d07dc243aac
Raspberry Pi Model:      Raspberry Pi 400 Rev 1.1
Architecture:            Linux raspberrypi 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [148        Thu Jan 27 21:04:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20220127-210418 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643313858 960 82 Northbound E
  * [145        Fri Jan 28 17:00:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20220128-170051 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643385651 868 23 Northbound E
  * [144        Fri Jan 28 08:51:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20220128-085128 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643356288 934 56 Southbound W
  * [166        Fri Jan 28 17:07:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20220128-170718 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643386038 900 36 Northbound E
  * [159        Fri Jan 28 18:49:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20220128-184913 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643392153 964 70 Northbound W
  * [165        Fri Jan 28 08:56:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20220128-085659 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643356619 937 33 Southbound W
  * [151        Fri Jan 28 12:39:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20220128-123951 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643369991 896 23 Southbound W
  * [150        Fri Jan 28 10:58:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20220128-105854 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643363934 1004 88 Southbound W
  * [167        Fri Jan 28 18:47:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20220128-184706 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643392026 922 51 Northbound W
  * [146        Fri Jan 28 18:39:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20220128-183917 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643391557 932 87 Northbound W
  * [163        Fri Jan 28 05:36:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20220128-053655 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643344615 768 11 Southbound E
  * [153        Fri Jan 28 22:34:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20220128-223403 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643405643 877 26 Northbound W
  * [164        Fri Jan 28 07:16:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M 2" METEOR-M-2-20220128-071635 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643350595 958 55 Southbound E
  * [152        Fri Jan 28 20:52:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20220128-205222 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643399542 959 68 Northbound E
  * [143        Fri Jan 28 07:11:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20220128-071137 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643350297 913 31 Southbound E
  * [156        Fri Jan 28 07:15:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20220128-071510 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643350510 918 33 Southbound E
  * [157        Fri Jan 28 08:55:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20220128-085554 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643356554 987 55 Southbound W
  * [149        Thu Jan 27 22:46:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20220127-224624 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643319984 875 22 Northbound W
  * [158        Fri Jan 28 17:09:00 2022 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20220128-170933 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1643386173 908 28 Northbound E
---------------------------------------------
Satellite SDR Settings:
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: 0
  noaa_15_enable_bias_tee: true
  noaa_15_gain: 40.2
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 20
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: 0
  noaa_18_enable_bias_tee: true
  noaa_18_gain: 40.2
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 20
[... 59 lines cut ...]
Product ID:             0x2838
Manufacturer:           Realtek
Product:                RTL2838UHIDIR
Serial number:          00000001
Serial number enabled:  yes
IR endpoint enabled:    yes
Remote wakeup enabled:  no
__________________________________________
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

colinluthier, 2022-01-28:
@mattues Can you tell me what OS you are using? This looks like a bug in sorting out Buster vs Bulls Eye.

mattues, 2022-01-28:
I’m using the bullseye version

Cadair, 2022-01-28:
baaaahhhgghh. I have a patch I am working on which will fix this, we are doing *terrible* things with our python dependencies and this looks like it's the upshot.

So I can try and reproduce this, are you using a completely clean bullseye install with the latest master?

Cadair, 2022-01-28:
what is the complete command you ran which broke it? (so I can run it :wink: )

mattues, 2022-01-28:
Hmm I think it just was “sudo update” and then “sudo upgrade”

mattues, 2022-01-28:
Numpy has been updated from 1.18 to 1.19 in my installation, strange because the latest version of numpy is 1.22.

Cadair, 2022-01-28:
you mean `sudo apt update; sudo apt upgrade`?

We don't use the latest version of numpy, but I can guess what's happened, I will try and fix it. If you re-run `./install_and_upgrade.sh` in the noaa directory it *should* fix it until next time you apt update.

mattues, 2022-01-28:
Yea exactly sudo apt update ; sudo apt upgrade.

mihajlo2003petkovic, 2023-10-04:
Hi! I've updated the RN2 so follow the upgrade guide. It uses the latest numpy and I didn't have problems on both Raspberry Pi nor my testing PC. I'll close this issue for now but if it persists, we'll reopen it.

closed 2023-10-04 by mihajlo2003petkovic