jekhokie/raspberry-noaa-v2 #515: Orange PI

betolley, 2022-02-26:
I managed to get Raspberry Noaa to run on a orange pi One running the latest Armbian.

Several of the dependencies don't exist and I had to compile then or install them by hand because they had a different name.  But it is now working.

betolley, 2022-02-28:
Meteor works.  what would case this with noaa?
[image]

[image]

Cadair, 2022-02-28:
Could you share what you had to install / change by hand for other people who might want to do this? Merging #108 is the first step on the path to supporting more platforms for this.

betolley, 2022-03-01:
How do I fix the audio sample rate?  I downloaded the file and ran through noaaapt and it was viewable.  I need to to work on wxtoimg.  I read the slat correction and it didn't really do anything unless I did it wrong.  

Click on Options, Disable PLL
Next, Click on Image, then Slant correction
Draw a line alone one of those groups of similarly-colored pixels left to right.
Accept the sample rate it shows, making sure to set only NOAA sampling frequency
Click Options, (un check) Disable PLL

Here is how I installed it:
I created a user called pi then
I used the directions and up to here it was the same:

./install_and_upgrade.sh

Then I updated ~/raspberry-noaa-v2/ansible/roles/common/tasks/dependencies.yml

```
- name: install dependencies
  become: yes
  apt:
    update_cache: yes
    state: present
    name:
      - at
      - bc
      - cmake
      - ffmpeg
      - gmic
      - gnuradio
      - gr-osmosdr
      - imagemagick
      - jq
      - libatlas-base-dev
      - libgfortran5
      #- libjpeg9
      #- libjpeg9-dev
      - libncurses5-dev
      - libncursesw5-dev
      - libsox-fmt-mp3
      - libusb-1.0-0
      - libusb-1.0-0-dev
      - libxft-dev
      - libxft2
      #- predict
      - python3-pip
      - python-setuptools
      - socat
      - sox
      - sqlite3

- name: install wkhtmltox (wkhtmltoimg)
  become: yes
  apt:
    deb: https://github.com/wkhtmltopdf/packaging/releases/download/0.12.6-1/wkhtmltox_0.12.6-1.raspberrypi.buster_armhf.deb
    state: present
```

Then i installed predict by hand from the predict website.  I had to alter it so that there was a link in /usr/bin for predict

Then I 
```
sudo apt-get install libjpeg9-dev
sudo apt-get install libjpeg62-turbo-dev libjpeg62-turbo
```

But wasn't sure if that was enough so I also downloaded jpegsr6.zip

wget https://versaweb.dl.sourceforge.net/project/libjpeg/libjpeg/6b/jpegsr6.zi

and compiled and installed it.

Then I manually installed these

```
sudo pip3 install setuptools
sudo pip3 install pyyaml
sudo pip3 install jsonschema
sudo apt-get install python3-ephem
sudo apt-get install python-numpy
sudo apt-get install python-matplotlib
sudo apt-get install python-pillow
sudo apt-get install python-pyyaml
sudo apt-get install python3-pyyaml
sudo apt-get install python3-pillow
sudo apt-get install python-yaml
sudo apt-get install python-yaml
sudo apt-get install python-ephem
sudo apt-get install python-idna
sudo apt-get install python-jsonschema
sudo apt-get install python-yaml
sudo apt-get install python-requests
```

And edited the requirements file to this:
```
#ephem==3.7.7.1
#idna==2.10
#jsonschema==3.2.0
#matplotlib==3.3.4
#numpy==1.19.5
#Pillow==8.1.2
#pyyaml==5.3.1
#requests==2.25.1
```

betolley, 2022-03-27:
I had to give up on the orange pi one with it.  I ended up buying a HP G1 Chromebox and changed the os to armbian and followed the same steps as above and it is working great now
[image]
.

closed 2022-03-27 by betolley