jekhokie/raspberry-noaa-v2 #517: [Error] NOAA images not working but METEOR images ar

oli123v, 2022-03-16:
Hello, as you can see in the attached images, the METEOR ones are pretty good, but the NOAA ones all look like the attached image. no idea why this is. 
[image]
[image]
[output.log](https://github.com/jekhokie/raspberry-noaa-v2/files/8269331/output.log)

betolley, 2022-05-05:
Take a look at this.  I don't have a answer for you.  https://github.com/jekhokie/raspberry-noaa-v2/issues/515

I was using a orange pi one with similar results.  When you listened to the audio it played too fast.  It caused NOAA
 issues when converting audio.  But METEOR was ok.    I ended up moving to a intel based Chromebox that I loaded with Armbian x86 and setup.  This is my most recent pass over with same hardware otherwise
[image]

.

mihajlo2003petkovic, 2023-10-04:
Try updating to the latest version and try using gnuradio or satdump_live for NOAA reception. This looks like image break up which happens sometimes with RTL_FM. That's the reason why I switched to gnuradio.

closed 2024-12-02 by MihajloPi