jekhokie/raspberry-noaa-v2 #521: [Error] git pull error

wifi75, 2022-03-29:
hello after send git pull comand for upgrate to latest version ,  there is thi error:

```
Update of 7badd2f..2cd1fd0
error: Your local changes to the following files would be overwritten with merge:
         ansible / roles / webserver / tasks / main.yml
         ansible / roles / webserver / templates / nginx_default.j2
         ansible / roles / webserver / templates / nginx_tls_default.j2
         config / settings_schema.json
         scripts / audio_processors / rtlsdr_m2_lrpt_rx.py
         scripts / audio_processors / rtlsdr_noaa_apt_rx.py
         scripts / image_processors / histogram.sh
Commit or stash your changes before you merge.
I abort the operation

```

how I can solve?

guidocioni, 2022-03-30:
As `git` is telling you, you modified some of the files, e.g. `main.yml`, and `git` is refusing to merge changes from the remote that will overwrite your local changes. 

You can either do a commit to preserve those changes or just do a stash to keep them for later. Either way you need to have a clean state of your source code before trying to pull changes from the remote locally.

mihajlo2003petkovic, 2023-10-03:
Make backup and delete this files and run git pull to update your system. After that, you can restore them and run the install script.

closed 2023-10-03 by mihajlo2003petkovic