jekhokie/raspberry-noaa-v2 #522: Installing on ubuntu is a nightmare

guidocioni, 2022-03-29:
I have already the pipeline from the instructable that this package is based on working on my raspi server running ubuntu so I thought I'd give this really cool package a shot! Unfortunately the sheer amount of things that I had to fix is destroying me, and still I cannot have a working installation. 

First of all, here is a list of all the things that I had to change for the installation (maybe it's useful for other people):

1. change the user everywhere in the code using the trick here https://github.com/jekhokie/raspberry-noaa-v2/issues/108#issuecomment-930699534, although this screws up some parts of the code where `pip` is obviously not the user :-) 
2. ignore the dependencies `libjpeg9` (-dev) as the 8 is available on ubuntu.
3. ignore the `predict` dependency as it was not finding the package on `apt`, although I already had it installed! There should be a check in the `yml` because there is no point in installing if it's already there 
4. Change `wkhtmltopdf` link to https://github.com/wkhtmltopdf/packaging/releases/download/0.12.6-1/wkhtmltox_0.12.6-1.focal_arm64.deb. I think there should be a check for the architecture at the beginning of the script which then get the right link: having an harcoded link maybe is not a good idea
5. Added a check to see if `wxtoimg` is already installed before attempting to do it in `tools.yml`, this was failing as in wkhtmltopdf although I already had it installed
6. Change php7 to php8 as it is not available anymore (why is 7 still used?) 

After doing that I was able to run the installer. However I'm still not able to reach the web interface. 
I have this as my `nginx` configuration file 

```
server {
    server_name ...;
    root /var/www/wx-new/public/;
    index index.php index.html index.htm index.nginx-debian.html;
    autoindex on;
    location ~ /images/(.*) {
        alias /srv/images/$1;
    }

    location ~ /videos/(.*) {
        alias /srv/videos/$1;
    }

    location / {
        try_files $uri $uri/ /index.php;
    }

    location ~ \.php$ {
        include snippets/fastcgi-php.conf;
        fastcgi_pass unix:/var/run/php/php8.0-fpm.sock;
    }

    # Rewrite the old path yyyy/mm/dd/ to /images/
    rewrite "^/image/\d{4}/\d{2}/\d{2}/(.+)$" /images/$1 permanent;

    listen 443 ssl; # managed by Certbot
    ssl_certificate ... # managed by Certbot
    ssl_certificate_key ....; # managed by Certbot
    include /etc/letsencrypt/options-ssl-nginx.conf; # managed by Certbot
    ssl_dhparam /etc/letsencrypt/ssl-dhparams.pem; # managed by Certbot
}

server {
    if ($host = ....) {
        return 301 https://$host$request_uri;
    } # managed by Certbot
    server_name ...;
    listen 80;
    return 404; # managed by Certbot
}
```

And if I try to reach the url I get 

```
Fatal error: Uncaught Exception: Unable to open database: unable to open database file in /var/www/wx-new/Lib/Controller.php:46 Stack trace: #0 /var/www/wx-new/Lib/Controller.php(46): SQLite3->__construct() #1 /var/www/wx-new/Lib/Controller.php(12): Lib\Controller->connectToDB() #2 /var/www/wx-new/Lib/Router.php(50): Lib\Controller->__construct() #3 /var/www/wx-new/Lib/Router.php(13): Lib\Router->handleRequest() #4 /var/www/wx-new/public/index.php(16): Lib\Router->__construct() #5 {main} thrown in /var/www/wx-new/Lib/Controller.php on line 46
```

Any idea what's causing this? Maybe it's related to the fact that I changed php7 to php8?

Did anyone succeed installing this on ubuntu? And did anyone manage to install the web interface into a sub-folder in nginx instead than being it the whole site? I already have nginx running on this server with other applications so it would be good being able to use it like that.

guidocioni, 2022-03-30:
I solved the `nginx` issue. Basically the Home folder (/home/) was lacking +x permissions for all users, and thus `php` could not read inside the `raspberry-noaa/db` folder although this folder already had +x permissions...took me a while to figure this out. 

If anyone has hints on how to put the web panel on a directory of the website with `nginx` please do let me know :)

i6cbi, 2022-03-31:
Hi Claudio, I have just finished installing the basic version 1.80 of RN2 on pc-I5 & linux-Mint with many difficulties due to my little knowledge of Linux.
While I'm still having trouble installing the new version of MeteorDemod on version V2.
If we want to talk about it ..
Greetings Silvio
also on https://discord.gg/hQzWXQpk

closed 2022-04-11 by guidocioni