jekhokie/raspberry-noaa-v2 #523: Running the web interface in a nginx sub-folder

guidocioni, 2022-03-30:
I already have `nginx` running on my server so I cannot host the application in the main root. 
For this reason I tried to change the `nginx` configuration to host the web interace into a sub-folder. 

```nginx
location /noaa {
    alias /var/www/wx-new/public/;
    try_files $uri $uri/ @noaa;
    index index.php index.html index.htm index.nginx-debian.html;
    autoindex on;
    location ~ /images/(.*) {
        alias /srv/images/$1;
    }

    location ~ /videos/(.*) {
        alias /srv/videos/$1;
    }

    location ~ \.php$ {
        include snippets/fastcgi-php.conf;
        fastcgi_param SCRIPT_FILENAME $request_filename;
        fastcgi_pass unix:/var/run/php/php8.0-fpm.sock;
    }
}

location @noaa {
    rewrite /noaa/(.*)$ /noaa/index.php?/$1 last;
}
```

and I'm almost there. index.php is found and served, together with the images and videos directory which now point to the correct sub-folder. However, the page doesn't load as expected, instead it prints 404 - could not find controller.
Looking at the code I can see where this is happening: basically `Router.php`, which is called in `index.php`, tries to construct a link to load files from the `App` folder but fails.

Does anyone know how to adjust the `nginx` configuration?

dom-robinson, 2022-03-30:
Hey Guido - couple of things:

are you editing the sites-available/default file? (rather than the nginx.conf file?)

second here is my sites-available for ref (on a pi - not shared nginx) - it might have some clues.>>

```                                                                                         
server {
  listen 8800;
  listen 443 ssl;
  root /var/www/wx-new/public/;
  index index.php index.html index.htm index.nginx-debian.html;
  autoindex on;

  server_name -----;

  ssl_certificate /etc/letsencrypt/live/---.----------.com/fullchain.pem;
  ssl_certificate_key /etc/letsencrypt/live/---.----------.com/privkey.pem;

  location ~ /images/(.*) {
    alias /srv/images/$1;
  }

  location ~ /videos/(.*) {
    alias /srv/videos/$1;
  }

  location / {
    try_files $uri $uri/ /index.php;
  }

  location ~ \.php$ {
    include snippets/fastcgi-php.conf;
    fastcgi_pass unix:/var/run/php/php7.4-fpm.sock;
  }

  # Rewrite the old path yyyy/mm/dd/ to /images/
  rewrite "^/image/\d{4}/\d{2}/\d{2}/(.+)$" /images/$1 permanent;
}

dom-robinson, 2022-03-30:
Theres a lot of folks in https://discord.gg/Jd7TZUtj for broader / faster help ..

guidocioni, 2022-03-30:
Thanks @dom-robinson, unfortunately this is part of the default configuration, which works for me if I want to run only the web interface of raspberry-noaa. However, I need to run multiple websites on different folder and that's why I need a slighlty different configuration which redirects all the traffic to e.g. `/noaa/`. It doesn't work because either
1. I'm doing something worng with `nginx`
2. The `php` website here is not written to support requests also from non-root envs

guidocioni, 2022-04-01:
> Theres a lot of folks in https://discord.gg/Jd7TZUtj for broader / faster help ..

If I'm not mistaken it says 'no technical support' 😜

guidocioni, 2022-04-10:
@jekhokie would it be possible to have a solution on this? That's the only thing preventing me from deploying it on the server. 
It's not clear who made the web interface but I suspect the solution is really simple.

I tried to run the server alone

`php -S 0.0.0.0:8010 -t /var/www/wx-new/public/`

but ofc got the same issue `Warning: Undefined array key "REQUEST_SCHEME" in /var/www/wx-new/Lib/Router.php on line 19`.

Somehow only with the right `nginx` configuration this can work, but I'm not sure how can I put this into a subfolder....
Another solution would be to serve this with a `lighttpd` instance, which I'm already running on this server, but also here I'm not sure how I would need to write the configuration to make the site work: just putting the root to `/var/www/wx-new/public/` does not work.

guidocioni, 2022-04-11:
So I solved it...after a weekend of wasted time. Unfortunately the `PHP` web interface is not really well written and does not really like to reside in anything but the root of the server. 

Suppose you want to run the website into a subfolder `/noaa/` in `nginx`.
First of all run the website standalone on a specific port in `nginx`: 

```nginx
server {
    root /var/www/wx-new/public/;
    index index.php index.html index.htm index.nginx-debian.html;
    autoindex on;
    location ~ /images/(.*) {
        alias /srv/images/$1;
    }

    location ~ /videos/(.*) {
        alias /srv/videos/$1;
    }

    location / {
        try_files $uri $uri/ /index.php;
    }

    location ~ \.php$ {
        include snippets/fastcgi-php.conf;
        fastcgi_pass unix:/var/run/php/php8.0-fpm.sock;
    }

    # Rewrite the old path yyyy/mm/dd/ to /images/
    rewrite "^/image/\d{4}/\d{2}/\d{2}/(.+)$" /images/$1 permanent;

    listen 8010; 
}
```

then in the main `nginx` website 

```nginx
location /noaa/ {
        proxy_pass http://127.0.0.1:8010;
        proxy_set_header Host $host;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
}
```
which was easy.... But then I had to change all the references in `/var/www/wx-new` to account for this new subfolder starting from `Router.php` by changing the indices of `$url`

```php
$this->controller = (isset($url[1]) and $url[1] != '') ? $url[1] : 'passes';
$this->action = (isset($url[2]) and $url[2] != '') ? $url[2] : 'index';
```

Then I had to change all the references in `Config.php`, that is 

```
const THUMB_PATH = '/noaa/images/thumb';
const ANIMATION_VIDEO_FILE = '/noaa/videos/RollingAnimation.mp4';
```

and finally I did a substitution of all the `href=""` and `src=""` in the `html` and `js` files in order to prepend `/noaa/` to all the links.

It would be good to rewrite the `php` website so that it's flexible and does not need to live in the root of the server.

closed 2022-04-11 by guidocioni