jekhokie/raspberry-noaa-v2 #524: Clarification on Time Zone setting

guidocioni, 2022-03-31:
My Raspberry is set on UTC as datetime even though I live in Germany which is UTC+1: it's just easier for me to handle everything when is set like that. 

How should I set the `timezone_offset` and `timezone: Europe/Berlin` parameters to use UTC in the backend but convert all the times (to be more readable) to local time in the frontend? And why there are two parameters if one can be clearly obtained from the other? :) 

Thanks

mihajlo2003petkovic, 2023-10-04:
I've removed the timezone settings as it was confusing any user, now we're pulling it directly from the Pis locale. That's in the latest update.

megabyte2010, 2024-06-25:
Sorry to re open this thread but how do I change this? The website is showing wrong time in passages. Lisbon should be +1.
[image]

MihajloPi, 2024-06-26:
Hello, did you correctly set locale in your Pi when installing your OS? Did you install RN2 on OS yourself or did you use a prebuild image? Was the incorrect timezone present from start or did it happen when daylight savings time took place?

megabyte2010, 2024-06-26:
Prebuilt image. I installed the image this week, so the problem is from start. System time is ok. 
[image]

megabyte2010, 2024-06-26:
[image]
[image]

creas002, 2024-06-26:
> Screenshot_20240626_100819_RVNC.Viewer.jpg (view on web)
> Screenshot_20240626_100743_RVNC.Viewer.jpg (view on web)

Can you run the following 3 commands from a terminal window and share the
results please?

*timedatectl*

*bash -x
$HOME/raspberry-noaa-v2/scripts/tools/configure_php_local_timezone.sh*

*cat $HOME/raspberry-noaa-v2/webpanel/public/index.php*

megabyte2010, 2024-06-26:
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
You have new mail.
Last login: Wed Jun 26 22:40:07 2024 from 192.168.1.222
nelsonfx@raspberrypi:~ $ timedatectl
               Local time: qua 2024-06-26 22:47:28 WEST
           Universal time: qua 2024-06-26 21:47:28 UTC
                 RTC time: n/a
                Time zone: Europe/Lisbon (WEST, +0100)
System clock synchronized: yes
              NTP service: n/a
          RTC in local TZ: no
nelsonfx@raspberrypi:~ $ bash -x $HOME/raspberry-noaa-v2/scripts/tools/configure_php_local_timezone.sh
bash: /home/nelsonfx/raspberry-noaa-v2/scripts/tools/configure_php_local_timezone.sh: Ficheiro ou pasta inexistente
nelsonfx@raspberrypi:~ $ cat $HOME/raspberry-noaa-v2/webpanel/public/index.php
<?php

# composer auto-loading
require dirname(__DIR__) . '/vendor/autoload.php';

use Config\Config;

# error handling
error_reporting(E_ALL);
ini_set('display_errors', 1);
ini_set('display_startup_errors', 1);

include(__DIR__ . '/../Lib/Router.php');

# handle route dispatching
$router = new Lib\Router();

?>
nelsonfx@raspberrypi:~ $

creas002, 2024-06-26:
@megabyte2010 
  The absence of this file --> **configure_php_local_timezone.sh**  tells me your image you installed is outdated.

I did not see any output from running this command **cat $HOME/raspberry-noaa-v2/webpanel/public/index.php** , so either the file is empty or you did not capture the output and share the results.  If the file was missing then there would have been an error when you ran the cat command. Please try to run the command again and paste results if different than last time. I would expect to see something. If its really empty then there you may have bigger problems. I've attached an index.php file that should resolve the issues you are having. If you use it then you will need to run install_and_upgrade.sh again.

# 1) Backup your existing index.php file 
cp -p $HOME/raspberry-noaa-v2/webpanel/public/index.php $HOME/raspberry-noaa-v2/webpanel/public/index.php.original

# 2) Using something like WinSCP or your favorite Linux text editor like VI or NANO - Copy the attached index.php file I uploaded to this comment over the top of  $HOME/raspberry-noaa-v2/webpanel/public/index.php

# 3) Execute install & upgrade script
cd $HOME/raspberry-noaa-v2/
./install_and_upgrade.sh

# 4) Check your webportal again to confirm if the timezone is now reporting correctly

[index.txt](https://github.com/user-attachments/files/15996978/index.txt)

megabyte2010, 2024-06-26:
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
You have new mail.
Last login: Wed Jun 26 22:47:21 2024 from 192.168.1.222
nelsonfx@raspberrypi:~ $ cat $HOME/raspberry-noaa-v2/webpanel/public/index.php
<?php

# composer auto-loading
require dirname(__DIR__) . '/vendor/autoload.php';

use Config\Config;

# error handling
error_reporting(E_ALL);
ini_set('display_errors', 1);
ini_set('display_startup_errors', 1);

include(__DIR__ . '/../Lib/Router.php');

# handle route dispatching
$router = new Lib\Router();

?>
nelsonfx@raspberrypi:~ $

megabyte2010, 2024-06-26:
nelsonfx@raspberrypi:~ $ cat $HOME/raspberry-noaa-v2/webpanel/public/index.php
<?php

# composer auto-loading
require dirname(__DIR__) . '/vendor/autoload.php';

use Config\Config;

# error handling
error_reporting(E_ALL);
ini_set('display_errors', 1);
ini_set('display_startup_errors', 1);

include(__DIR__ . '/../Lib/Router.php');

# handle route dispatching
$router = new Lib\Router();

?>
nelsonfx@raspberrypi:~ $

megabyte2010, 2024-06-26:
<?php

# composer auto-loading
require dirname(__DIR__) . '/vendor/autoload.php';

use Config\Config;

# error handling
error_reporting(E_ALL);
ini_set('display_errors', 1);
ini_set('display_startup_errors', 1);

include(__DIR__ . '/../Lib/Router.php');

# handle route dispatching
$router = new Lib\Router();

?>

creas002, 2024-06-26:
Back it up and try

<?php

date_default_timezone_set('Europe/Lisbon');

# composer auto-loading
require dirname(__DIR__) . '/vendor/autoload.php';

use Config\Config;

# error handling
error_reporting(E_ALL);
ini_set('display_errors', 1);
ini_set('display_startup_errors', 1);

include(__DIR__ . '/../Lib/Router.php');

# handle route dispatching
$router = new Lib\Router();

?>

Then Execute install & upgrade script
cd $HOME/raspberry-noaa-v2/
./install_and_upgrade.sh

megabyte2010, 2024-06-26:
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
You have new mail.
Last login: Wed Jun 26 22:47:21 2024 from 192.168.1.222
nelsonfx@raspberrypi:~ $ cat $HOME/raspberry-noaa-v2/webpanel/public/index.php
#I can't paste this here without this##<?php

# composer auto-loading
require dirname(__DIR__) . '/vendor/autoload.php';

use Config\Config;

# error handling
error_reporting(E_ALL);
ini_set('display_errors', 1);
ini_set('display_startup_errors', 1);

include(__DIR__ . '/../Lib/Router.php');

# handle route dispatching
$router = new Lib\Router();

?>
nelsonfx@raspberrypi:~ $

creas002, 2024-06-26:
github is trashing our comments.  lol

creas002, 2024-06-26:
This is the key line you need to add to your file

##<?php

date_default_timezone_set('Europe/Lisbon');

megabyte2010, 2024-06-26:
Ok it's done, but the software shouldn't get the correct time without adding a patch?

[image]

creas002, 2024-06-26:
Were you not trying to make it report UTC+01 opposed to UTC+02?  Your last screenshot looks like its working to me.  You may want to reschedule passes and check if the time look correct.

megabyte2010, 2024-06-26:
I added the patch to index. Then I ran the script. It's showing the correct time now like the print shows. My question is, the software shouldn't be able to get the correct time without adding a patch to index?

creas002, 2024-06-26:
Yes, if you had the latest version of software installed from github using git clone commends as the README states, but you are using an image someone built and does not have the fixes we have made.

megabyte2010, 2024-06-26:
It's because this isn't the first time I run this project. Maybe in 2022 I built a ground station with raspinoaa and this exact problem was there.

megabyte2010, 2024-06-26:
I got the image from here

https://qsl.net/ve3elb/RaspiNOAA/

creas002, 2024-06-26:
I started helping support RN2 within the last month. Prior to that I was like yourself using images someone else built. I got tired of battling issues, so I decided to learn a bit how it is all glued together. I made it work on Bookworm 12 so RPi5 could be supported. Along the way I saw a number of things that needed fixing, so this was one of those things. I'm not the author or even a real supporter of RN2 but quickly starting to step into that role. I am not surprise that the issue you are reporting has been around a long time. There are a lot of moving pieces under the hood and just a handful of talented folks to work on things and they have lives as well. So, I stopped depending on someone else's images and went to RN2 and not VE3ELB image that I believe is bundled up by someone else. Honestly I deleted my previous comment because I was afraid of leading you down the wrong path because I just don't know whats in that image or even that it originated from RN2. Well, I'm glad you are working correctly now. @MihajloPi can mark this issue closed again. Personally I believe it would make more sense to ask the person whom built the image for support because they know how it was built and limits or things needing to be address.

megabyte2010, 2024-06-26:
don´t worry. its working now. i tried to see if the noaa 18 is where the software is saying and it is. so its ok. i asked here because this page says that there is prebuilt images in that link. good work.

ve3elb, 2024-06-27:
@creas002 thank you for helping out the group. 

In reply to your comment:
"I stopped depending on someone else's images and went to RN2 and not VE3ELB image that I believe is bundled up by someone else. Honestly I deleted my previous comments because I was afraid of leading you down the wrong path because I just don't know whats in that image or even that it originated from RN2. Well, I'm glad you are working correctly now."

If you read the git page the VE3ELB Image is the official RN2 image and it is complied from the latest RN2 git page. The image is clean.  This image has been around for quite a while and is updated frequently. As for issues it will have the same issues as git and most times they are all fixed in the image before release.  The most common issue everyone is having is with timezone and passes time not being correct. This was fixed last year but some of the users do not run the schedule command after setup so things don't work. There is a set up instruction pdf file but some still seem to skip ahead of some steps. 

""I started helping support RN2 within the last month. Prior to that I was like yourself using images someone else built. I got tired of battling issues, so I decided to learn a bit how it is all glued together.""

In any case whatever help you have given with RN2 we all greatly appreciate it. The VE3ELB Image is the same as RN2 GitHub just in a P&P form for the users who are not familiar with Linux or just don't have the time to compline it themselves.

@MihajloPi Deals with RN2 GitHub and I deal with RaspiNOAA V2, VE3ELB Image so everyone can enjoy the hobby. 

Support for RN2 git and the VE3ELB Image is the same.  

And as always thanks to everyone for supporting RN2/RaspiNOAA V2

creas002, 2024-06-27:
Understood, if you read the README
https://github.com/jekhokie/raspberry-noaa-v2 it also says

" These images are not always up to speed with the latest code, but lots of
folks find images are a great way to get started quickly!"

megabyte2010, 2024-06-27:
Just to add that my problem was only with the timezone displayed in website. I ran the schedule script and the passes was right but displayed with one hour ahead in relation with the local time.

MihajloPi, 2024-06-27:
I probably should add to the README page that sometimes updating a pre-built image can break certain things in RN2 so it's better not to update the image with `git pull` itself, but wait for another image release and flash it on the SD card. I've already had one user who had updated VE3ELB's image and had issues with the website. It's best to wait for the image author's to update their images since the Github developers cannot keep with all different image versions, because they work on the current leading Github version directly. If user's want to update their RN2 versions frequently and keep up with all the new features or even be beta testers, they should install OS themselves and follow the Github instructions to install RN2 directly as it's very easy now. If users want to have the convenience of using the image, beware of potential breakages when updating as every author's image is a bit different. I've built an image myself too in September 2023 and I still don't recommend users who wish to run up-to-date setup download my image and run `git pull` because I am certain they'll have much harder time then if they were to install RN2 with the Github instructions themselves.

With that said, using image is a valid way of running a RN2 setup, but keep in mind that the Github developers can't give proper support for them, and updating should be done with the next image author's release. I'm closing this issue.

closed 2024-06-27 by MihajloPi