jekhokie/raspberry-noaa-v2 #528: MeteorGIS integration

Cornmeister, 2022-04-11:
Is it possible to add MeteorGIS integration ?
http://www.meteorgis.space/

I like the option to add country borders for Meteor too.

dom-robinson, 2022-04-12:
Feel free to try out this branch - [https://github.com/jekhokie/raspberry-noaa-v2/tree/meteordemod-staging](https://github.com/jekhokie/raspberry-noaa-v2/tree/meteordemod-staging) 

we are planning to merge it soon. It uses meteordemod and adds maps. Example output included.
[image]
[image]

dom-robinson, 2022-04-27:
oh thats a shame - its definately workable on a clean bullseye

Cornmeister, 2022-04-28:
> oh thats a shame - its definately workable on a clean bullseye

Yes, it works fine on a clean install.

mihajlo2003petkovic, 2023-10-03:
It works by default now. Try updating your RN2 to the latest version and reinstall it. Meteor_receiver should be set to gnuradio.