jekhokie/raspberry-noaa-v2 #534: Including SDRplay on the image.

LeFEAR, 2022-04-25:
**Please Describe the problem to be solved**
I had this issue with my setup that I have only an SDRplay in my possession. Due to the Chip shortage i couldn't source an RTLSDR for any normal amount of money. (and at a reasonable time frame )

So I think i managed to make it work with SDRplay but I need a bit of your help , to know if I made any mistakes. 

So I downloaded the TCP image from SDRplay and burned it onto the rpi. 

Then I build from source your R-NOAA-V2.

Now the tricky part, I installed rx_fm from source, 

and now I replaced in the audio processors all parts that wore pointing to RTL_FM and removed the bias_t options / device select. 

at the moment I am not sure if I am missing anything that could be potentially a problem like bandwidth option .. or something like this. 

I am sure someone like me will be in need of this "mod" 

if you can refine it and make sure that it works as expected ( because I am no linux guru by any means of the imagination ) 

I think people would be grateful as I am :D I will wait to see if I will decode anything later today :D

dom-robinson, 2022-04-25:
let us know how you get on. some time ago we added the option to develop the capture workflow using gnuradio mode. That provides a fairly well (community) supported environment which opens the workflow up to hardware other than RTLSDRs - although the version currently available was included with only RTLSDR setup - there is more on it here: [https://github.com/jekhokie/raspberry-noaa-v2/blob/master/scripts/audio_processors/README.md](https://github.com/jekhokie/raspberry-noaa-v2/blob/master/scripts/audio_processors/README.md)

That would be the more widely supported way to add SDRPlay - but that said I dont know anyone who has done so yet! If your mods work then do share them here for those interested!

LeFEAR, 2022-04-25:
well it didn't work for some reason.. :'( 
Thanks for the help :D

dom-robinson, 2022-04-26:
Hey no problem :) 

Have a play with this example line to manually trigger >> [https://github.com/jekhokie/raspberry-noaa-v2/blob/2cd1fd096fdf308093c994c3a3d0b203f9467db4/scripts/receive_meteor.sh#L15](https://github.com/jekhokie/raspberry-noaa-v2/blob/2cd1fd096fdf308093c994c3a3d0b203f9467db4/scripts/receive_meteor.sh#L15)

LeFEAR, 2022-04-27:
> Hey no problem :)
> 
> Have a play with this example line to manually trigger >>
> 
> https://github.com/jekhokie/raspberry-noaa-v2/blob/2cd1fd096fdf308093c994c3a3d0b203f9467db4/scripts/receive_meteor.sh#L15
> 
> And note there is a similar one in the receive_noaa.sh script

Yes I did find that line .. and I used it and it worked... 

But when the schedule went to execute the recording didn't happen ... 

And the logs are not detailed enough to see why the script taps out ..

wifi75, 2022-04-29:
Hi, could someone explain to me how to listen a passage in real time?

LeFEAR, 2022-04-29:
> Hi, someone could explain to me how to listen a passage in real time?

You can find the recordings in the following path to play back the audio : 
https://github.com/jekhokie/raspberry-noaa-v2/blob/master/docs/troubleshooting.md#images-and-audio

Images are stored in the **/srv/images** directory, and audio in **/srv/audio.** These directories are opened for access by the webpanel to display in the browser.

As far as I know live decode is not possible. This image first records the audio /data and then goes to decode. 

Thanks :)

LeFEAR, 2022-04-29:
> Hey no problem :)
> 
> Have a play with this example line to manually trigger >>
> 
> https://github.com/jekhokie/raspberry-noaa-v2/blob/2cd1fd096fdf308093c994c3a3d0b203f9467db4/scripts/receive_meteor.sh#L15
> 
> And note there is a similar one in the receive_noaa.sh script

I managed to eliminate what ever issues / error that may hang up the processing, 

but when I dry run the script I get this ... 

./receive_noaa.sh: line 209: 11259 Segmentation fault      $WXMAP -T "${SAT_NAME}" -H "${TLE_FILE}" -p 0 ${extra_map_opts} -o "${epoch_adjusted}" $map_overlay >> $NOAA_LOG 2>&1

Thanks

LeFEAR, 2022-05-01:
Okay it worked. Now it records the audio. and decodes. 

--> What I had to go to change the path of`RTL_FM="/usr/local/bin/RTL_FM" -->`RTL_FM="/usr/local/bin/**rx_fm**"
`
in the common.sh file. 
and before that to build the rx_fm and copy the rx_fm binary into the /usr/local/bin/ directory so that you can run that program from anywhere. 
and that was it. it worked. 
the other issue I encountered was the BIAS T command for some reason that was giving an error on the rx_fm and I just had to remove that atribute from the audio processors part of the command. 

But there are a lot of issues with gain and IF and RF gain .. because they are special snowflakes.  vs RTLSDR. 

Now next question is to see why I get no image on the decode. 

most probably something going on with the Gain details. I would like to know if there is a way to save the output of the rx_fm into a file ? 

that output that gets posted on the terminal when you run it manually. 

Thanks for the help.

dom-robinson, 2022-06-30:
Sorry - been busy / away - looks like you made some good progress!

LeFEAR, 2022-06-30:
> Sorry - been busy / away - looks like you made some good progress!

Hello, thank you for getting back to me,

Unfortunately something happened with the complex gains SRPLAY has... And i was unable to get anything.

There are a great confusion with all the different gains sdrplay has and at some point i feel i fried my sdr... With all the gain i added to it.

So unfortunately I am unable to continue testing it.

Even using a USB host / client server i was unabke on Windows to make it work properly so its safe to assume that my sdr is dead. RIP.

I will bite to bullet and buy an Rtlsdr and call it a day. BUT still i am very disappointed with sdrplay. 

Thank you in advance

mihajlo2003petkovic, 2023-10-04:
Hi! satdump_live supports SDR Play. All you need is to install the API on your system. Follow the upgrade guide from Github on your setup to update your system.