jekhokie/raspberry-noaa-v2 #535: Meteor M2 not show

wifi75, 2022-04-27:
Hello on my passes list meteror m2 is present.
after the passage even after a few hours I do not see any image of the passage of meteor m2, the tab does not even appear, yet in the configuration it is enabled.

why?

dom-robinson, 2022-04-27:
does the output.log not detail what is happening?

wifi75, 2022-04-27:
I am attaching my log, I am not able to understand ...
[output.log](https://github.com/jekhokie/raspberry-noaa-v2/files/8570901/output.log)

guidocioni, 2022-05-06:
Been seeing the same issue today.
I usually get always a METEOR M2 image but since today I'm not able to get any results.

The recording seems fine (according to the spectogram) 
[image]

but then in the decoding it doesn't seem to find any valid package, as in @wifi75. May be the same issue

guidocioni, 2022-05-07:
Apparently METEOR M2 has been having some positional issues and the VHF stream has been disabled since 4 May. 
This is the reason behind the missing images.

dom-robinson, 2022-06-30:
This this was satellite side - closing this now

closed 2022-06-30 by dom-robinson