jekhokie/raspberry-noaa-v2 #536: update git

wifi75, 2022-04-29:
hello after send git pull show this below:

```
 hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Updating 7badd2f..2cd1fd0
error: Your local changes to the following files would be overwritten by merge:
        ansible/roles/webserver/tasks/main.yml
        ansible/roles/webserver/templates/nginx_default.j2
        ansible/roles/webserver/templates/nginx_tls_default.j2
        config/settings_schema.json
        scripts/audio_processors/rtlsdr_m2_lrpt_rx.py
        scripts/audio_processors/rtlsdr_noaa_apt_rx.py
        scripts/image_processors/histogram.sh
Please commit your changes or stash them before you merge.
Aborting
```
what can I do for update ?

dom-robinson, 2022-05-03:
I would do a clean install, and then once you do the pull do ```git checkout meteordemod-staging```

Then carefully follow the read me on [https://github.com/jekhokie/raspberry-noaa-v2/blob/meteordemod-staging/README.md](https://github.com/jekhokie/raspberry-noaa-v2/blob/meteordemod-staging/README.md)

and grab that version. There is even a more improved version called [Build-MeteorDemod-from-upstream](Build-MeteorDemod-from-upstream) However i am not personally familiar with it. I do know it is trying to get the meteordemod-staging into a state to promote to master. 

You might find that to be the cleanest and most fully featured set of options since that is where most work is at the moment.

dom-robinson, 2022-06-30:
Did that work? Can we close this one?

mihajlo2003petkovic, 2023-10-04:
Hi! Follow the upgrade guide on Github, it has been updated.