jekhokie/raspberry-noaa-v2 #537: [Error] running install_and_update get fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'libjpeg9' is available"}

surfrod, 2022-05-11:
while doing a new install on a new raspberry pi 4, I get a fatal error:
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'libjpeg9' is available"}

Tried installing 
sudo apt-get install libjpeg-dev

But still getting the error

surfrod, 2022-05-12:
ok, I changed the dependencies on libjpeg9 to libjpeg62 which is found in the distro.

In addition, since I don't know ansible, I removed the dependency on predict, and found kd2bd version on git, downloaded and built it.  It seems the install of v2 succeeded and I now have passes scheduled for later today, 5/12.

If anyone knows how to update the files in the distro, probably a good idea to do that for bullseye going forward, if possible.

Cadair, 2022-05-14:
I thought we had already made this work, are you on the 32bit latest raspbian image?

surfrod, 2022-05-14:
No, I was on 64 bit. I just did a new install with a 32 bit package... too many headaches with Bullseye 64 bit.

I think this can be closed, unless someone wants to keep pressing on with the 64 bit version. I gave up trying to get wxtoimg to work. I actually got past the build issues with raspberry-noaa-v2 but could not get a version of wxtoimg that would work. Library incompatibilities. I think on discord someone said they got it working but I had fallen back to the 32 bit distro already (which is now installed, now to get v2 installed...)

Cadair, 2022-05-15:
Ok yeah, we don't support the 64bit image, that's being tracked in #463.

We really should add a note to the docs making it clear that you need the 32bit version.

closed 2022-05-15 by Cadair