jekhokie/raspberry-noaa-v2 #543: Problem woth NOAA images after 6-7 months

golvellius, 2022-06-30:
Hi . 

First of all i would like to thankyou for your supperb program. I am using a raspberrypi 2B. The program works perfect but since 8 months it start to work with some errors. Before i have good quality images and suddenly i have this small images and double images. I have tested the antenna and it's working, also i have change BIAS from 0-50 and the same results. Change antenna possitioning with no results. 
Also i have reinstalled the software with no luck.

[image]
[image]
[image]

What could be the problem?  Cable?

Thanks for all.

dom-robinson, 2022-06-30:
Can you check clocks and your lat / lon settings? If they are out then it will try to record when there is no satellite and record noise - these captures you have above are what i would expect with a pure-noise source so if you are confident the SDR and antenna are workign ok then it would indicate the timing might be out...

mihajlo2003petkovic, 2023-10-04:
@golvellius This shouldn't happen. My Pi has been running stable for 2 years now.