jekhokie/raspberry-noaa-v2 #545: [feature?] WXtoImg image generation

ShianMull, 2022-07-13:
Hi, huge thanks for this interesting project I’ve just started to get working.  I previously had an old Maplin (Radio Shack) VHF satellite decoder I built in the ‘90s.  SDR has opened a whole new none-knob-twiddling world for me!  I just had a question that WXtoImg PC version could produce combined orthographic projection images and an animated video through the day.  Is this possible in the Linux version you use here?  All the best, Mark

dom-robinson, 2022-07-13:
Hey Mark - I did have a little crack at the animation process. note [https://github.com/jekhokie/raspberry-noaa-v2/issues/347](https://github.com/jekhokie/raspberry-noaa-v2/issues/347)

If you turn on AVI it will composite each image as a new frame, but i have yet to find a way to align them like you can in the pc app..

ShianMull, 2022-07-13:
Thanks Dom,

I read the note with interest.  I think the only way to overlay on a consistent grid is to use spherical trig to overlay the scan strips on a defined-size square centred on users location.  You then have a frame that gives a consistent orientation for an AVI.

The spherical trig isn’t too difficult, just requires accurate start/stop recording time to position accurately - gives start long/lat and end long/lat, whilst the track angle gives the projected chord across the frame (the Earth’s diameter being a constant).

I think the actual software algorithm is already sorted and available in the final beta release of WXtoImg for Windows - I must have a look.

I wasn’t sure if you were using WXtoImg here, but it perhaps isn’t as comprehensive as the PC version?

all the best

Mark