jekhokie/raspberry-noaa-v2 #572: 404 Not Found

edegraaff, 2022-12-20:
**Describe the bug**
the *polar-direction.png file is not created for some reason

**Screenshots and/or logs**
[image]

**Support Output**
=============================================
Details about environment
=============================================
Current local date/time: Tue 20 Dec 22:16:26 CET 2022
Current date/time (ms):  1671570986
Repo git hash:           477b585800fb91f8c3c972851d3c0ce1faf4aa7c
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.5
Architecture:            Linux noaa 5.15.76-v7l+ #1597 SMP Fri Nov 4 12:14:58 GMT 2022 armv7l GNU/Linux
Num CPUs:                4
>> if needed i can add lots more

**Server information (please complete the following information):**
 - OS: rasbian bullseye lite 32 bit
 - Python version Python 3.9.2
 - SDR device RTL-SDR v3

**Additional context**
Brand new clean install executed today. Install basics with sudo or su and installed the complete package under user pi 

Polar direction image tile is not found, it is for some reason not created, no clue why not, all the other images are there. 
Not a big deal but maybe something is missing in the installer?

edegraaff, 2022-12-21:
Searched the logfile and find this one: 
  libpostproc    55.  7.100 / 55.  7.100
Input #0, png_pipe, from '/home/pi/raspberry-noaa-v2/tmp/temp_spectrogram.png':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: png, rgb24(pc), 652x928 [SAR 1:1 DAR 163:232], 25 tbr, 25 tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (png (native) -> png (native))
Press [q] to stop, [?] for help
[Parsed_drawtext_0 @ 0x18a5a90] Using "/usr/share/fonts/truetype/ttf-bitstream-vera/Vera.ttf"
Output #0, image2, to '/srv/images/NOAA-15-20221220-174108-spectrogram.png':
  Metadata:
    encoder         : Lavf58.45.100
    Stream #0:0: Video: png, rgb24, 652x928 [SAR 1:1 DAR 163:232], q=2-31, 200 kb/s, 25 fps, 25 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.91.100 png
frame=    1 fps=0.0 q=-0.0 Lsize=N/A time=00:00:00.04 bitrate=N/A speed=0.146x
video:295kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
20-12-2022 17:54 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Producing polar graph of direction for pass
RuntimeError: module compiled against API version 0xe but this version of numpy is 0xd
Traceback (most recent call last):
  File "/home/pi/raspberry-noaa-v2/scripts/image_processors/polar_plot.py", line 31, in <module>
    import matplotlib.pyplot as plt
  File "/usr/local/lib/python3.9/dist-packages/matplotlib/__init__.py", line 174, in <module>
    _check_versions()
  File "/usr/local/lib/python3.9/dist-packages/matplotlib/__init__.py", line 159, in _check_versions
    from . import ft2font
ImportError: numpy.core.multiarray failed to import
convert: unable to open image `/srv/images/NOAA-15-20221220-174108-polar-direction.png': No such file or directory @ error/blob.c/OpenBlob/2924.
convert: no images defined `/srv/images/thumb/NOAA-15-20221220-174108-polar-direction.png' @ error/convert.c/ConvertImageCommand/3229.
20-12-2022 17:54 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Bulding pass map
wxmap: warning: could not find matching pass to build overlay map.
20-12-2022 17:54 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Decoding image
wxtoimg: error: could not open overlay map file '/home/pi/raspberry-noaa-v2/tmp/map/NOAA-15-20221220-174108-map.png'
Satellite: NOAA
Status: signal processing............................
wxtoimg: warning: couldn't find telemetry data
Gain: 10.8
Channel A: 3/3B (mid infrared)
Channel B: 4 (thermal infrared)
Loading page (1/2)

mihajlo2003petkovic, 2023-10-04:
This should work natively, try updating your RN2. I've close this issue for now and we'll reopen it if it persists after the update. This is how it looks on my website:
[image]

closed 2023-10-04 by mihajlo2003petkovic