jekhokie/raspberry-noaa-v2 #574: [Error] No meteor image produced except spectrogram

gamani, 2023-01-22:
**Describe the bug**
The script is failing saying there is no image in the directory.
Only the spectrogram image can be produced.
However NOAA images are fine.

**Screenshots and/or logs**
```
22-01-2023 08:33 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : I got a successful METEOR-M-2-20230122-081155.dec file. Decoding APID 68
Reading /srv/audio/meteor/METEOR-M-2-20230122-081155.dec...
 pos=       0 (  0.00%) 
 pos=    1024 (  7.14%) 
 pos=    2048 ( 14.29%) 
 pos=    3072 ( 21.43%) 
 pos=    4096 ( 28.57%) 
 pos=    5120 ( 35.71%) 
 pos=    6144 ( 42.86%) 
 pos=    7168 ( 50.00%) 
 pos=    8192 ( 57.14%) 
 pos=    9216 ( 64.29%) 
 pos=   10240 ( 71.43%) 
 pos=   11264 ( 78.57%) 
 pos=   12288 ( 85.71%) 
 pos=   13312 ( 92.86%) 

Total:        0.003451
Processing:   0.000165
Correlation:  0.000000
Viterbi:      0.000000
ECC:          0.000000
Remainder:    0.003286
Packets:      14 / 14
convert: unable to open image `/srv/images/METEOR-M-2-20230122-081155-122.bmp': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/srv/images/METEOR-M-2-20230122-081155-122.bmp' @ error/convert.c/ConvertImageCommand/3258.
22-01-2023 08:33 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Rectifying image to adjust aspect ratio
Traceback (most recent call last):
  File "/home/pi/raspberry-noaa-v2/scripts/image_processors/meteor_rectify.py", line 121, in <module>
    img = Image.open(sys.argv[1])
  File "/usr/local/lib/python3.7/dist-packages/PIL/Image.py", line 2904, in open
    fp = builtins.open(filename, "rb")
FileNotFoundError: [Errno 2] No such file or directory: '/srv/images/METEOR-M-2-20230122-081155-122.bmp'
22-01-2023 08:33 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Annotating images and creating thumbnails
Loading page (1/2)
[>                                                           ] 0%
[======>                                                     ] 10%
[=======>                                                    ] 13%
[=====================>                                      ] 35%
[============================================================] 100%
Rendering (2/2)                                                    
[>                                                           ] 0%
[===============>                                            ] 25%
[============================================================] 100%
Done                                                               
convert: unable to open image `/srv/images/METEOR-M-2-20230122-081155-122-rectified.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: ImageSequenceRequired `-composite' @ error/mogrify.c/MogrifyImageList/7995.
convert: no images defined `/srv/images/METEOR-M-2-20230122-081155-122-rectified.jpg' @ error/convert.c/ConvertImageCommand/3258.
convert: unable to open image `/srv/images/METEOR-M-2-20230122-081155-122-rectified.jpg': No such file or directory @ error/blob.c/OpenBlob/2874.
convert: no images defined `/srv/images/thumb/METEOR-M-2-20230122-081155-122-rectified.jpg' @ error/convert.c/ConvertImageCommand/3258.
22-01-2023 08:33 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : No image produced - not pushing anywhere
22-01-2023 08:33 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Total processing time: 00:22.35
```

**Support Output**
Paste the output of running the `./support.sh` script here.

**Server information (please complete the following information):**
 - OS: Raspbian i386
 - Python version 2.7.3
 - SDR device RTL-SDR v3

KiwiInNZ, 2023-01-22:
Meteor M-2 has been offline since the 24th of December after our started
tumbling, which is why you'll be getting no images from it. Hopefully it
can be recovered.

gamani, 2023-01-22:
Thanks for the info. But here it should produce a blank image then ? The .dec file has been generated but afterwards it seems tihs line in the code is failing.
`      log "I got a successful ${FILENAME_BASE}.dec file. Decoding APID 68" "INFO"
      ${IMAGE_PROC_DIR}/meteor_apid68_decode.sh "${AUDIO_FILE_BASE}.dec" "${IMAGE_FILE_BASE}-122" >> $NOAA_L$
      $CONVERT $FLIP -negate "${IMAGE_FILE_BASE}-122.bmp" "${IMAGE_FILE_BASE}-122.bmp" >> $NOAA_LOG 2>&1`

gamani, 2023-01-22:
The M2 sat came right away onver me and it's visible on the spectrogram below. I would be really happy to fix the generation bug upper or at least to generate the images from the wav or dec file manually. Can somebody help here ? 
[image]

mihajlo2003petkovic, 2023-10-04:
Meteor M2 is dead. Try updating your RN2 by deleting settings.yml file, running git pull, fill out your new settings.yml and run the install script. You'll see new settings for M2-3 inside the settings.yml file.