jekhokie/raspberry-noaa-v2 #582: Captures no showing on webpage

Dentonknifeworks, 2023-04-08:
Everything has been working for over a year.
But now I show no captures on the webpage.
I cleared out the images and thumbnails
Got a day's worth of captures with no issues there have images and thumbnails on the PI.
It is like the PHP is not loading the images.
Need some ideas

nxugget, 2024-08-08:
I have the same issue, some images not loading but if I click on it, images does appears  : 
[image]

nxugget, 2024-08-08:
Thanks for the reply, I'm already using the **beta-development** and image generated are indeed `.png`. 
And it's weird that it only causes this problem for certain captures no ?

creas002, 2024-08-08:
It looked like your M2 captures were working and the NOAA was not or at lest the thumbnails we not. So it might be something in receive_noaa.sh script.  If you would placate me by running :

`${HOME}/raspberry-noaa-v2/scripts/tools/verification_tool/verification.sh quick`

and share log file located at

`/var/log/raspberry-noaa-v2/verification.log`

Thx

creas002, 2024-08-08:
Not sure when you cloned beta-development, but if it was more than a few days ago, you probably should

`cd $HOME/raspberry-noaa-v2`
`git pull`

To make sure you have the latest updates. 

Can you share the output of

`cat ${HOME}/raspberry-noaa-v2/.git/config `

creas002, 2024-08-08:
My thumbnail images on the webportal are type jpg and are created in /srv/images/thumb directory as jpg, except for polar-direction.png's  
[image]

nxugget, 2024-08-11:
Actually, like in my file `settings.yml` I have this : 
`noaa_daytime_enhancements: 'MCIR MCIR-precip therm'`
It didn't work because I don't have **MSA** captures.
 So I changed `webpanel/App/Views/Captures/index.html` by replacing this piece of code : 
```html
{% elseif capture.sat_type == 1 %}
   {% if capture.daylight_pass == 1 %}
      {% set image_path = constant('Config\\Config::THUMB_PATH') ~ '/' ~ capture.file_path ~ '-MSA.jpg' %}
      <img class="card-img-top" src="{{ image_path }}">
   {% else %}
       {% set image_path = constant('Config\\Config::THUMB_PATH') ~ '/' ~ capture.file_path ~ '-MCIR.jpg' %}
       <img class="card-img-top" src="{{ image_path }}">
    {% endif %}
```
to juste this : 
```html
{% elseif capture.sat_type == 1 %}
   {% set image_path = constant('Config\\Config::THUMB_PATH') ~ '/' ~ capture.file_path ~ '-MCIR.jpg' %}
   <img class="card-img-top" src="{{ image_path }}">
```

And now it works perfectly :) 
I don't know if this is the best solution but at least it works.