jekhokie/raspberry-noaa-v2 #586: Mystery...Spectrogram Displaying Wrong Frequency?

dandspach, 2023-06-22:
**Describe the bug**
I believe Spectrogram is showing incorrect frequency for NOAA sats.  Example:  NOAA 19 show decode at 2550MHZ.

**Screenshots and/or logs**
[image]

[image]

**Support Output**
=============================================
Details about environment
=============================================
Current local date/time: Wed 21 Jun 2023 11:10:04 PM EDT
Current date/time (ms):  1687403404
Repo git hash:           5ba567a3b5dfaac6a0c72dd0c25769c387092d31
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.2
Architecture:            Linux RaspiNOAA 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [1857	Fri Jun 23 12:16:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20230623-121639 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687536999 988 83 Southbound W
  * [1851	Sat Jun 24 20:28:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20230624-202807 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687652887 921 77 Northbound W
  * [1868	Fri Jun 23 21:21:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20230623-212136 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687569696 985 78 Northbound E
  * [1861	Sun Jun 25 11:52:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20230625-115218 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687708338 969 58 Southbound E
  * [1854	Tue Jun 27 09:35:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20230627-093517 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687872917 916 51 Southbound W
  * [1870	Sun Jun 25 20:57:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20230625-205751 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687741071 977 46 Northbound E
  * [1858	Fri Jun 23 23:39:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20230623-233935 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687577975 969 62 Northbound W
  * [1869	Sat Jun 24 21:09:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20230624-210941 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687655381 981 60 Northbound E
  * [1852	Sun Jun 25 08:46:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20230625-084618 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687697178 925 53 Southbound E
  * [1865	Thu Jun 22 10:10:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20230622-101049 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687443049 991 74 Southbound E
  * [1867	Fri Jun 23 09:58:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20230623-095851 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687528731 986 57 Southbound E
  * [1853	Sun Jun 25 20:02:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20230625-200250 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687737770 955 59 Northbound E
  * [1863	Mon Jun 26 23:03:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20230626-230303 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687834983 932 59 Northbound E
  * [1871	Tue Jun 27 10:51:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20230627-105131 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687877491 941 46 Southbound W
  * [1860	Sat Jun 24 23:27:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20230624-232720 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687663640 959 81 Northbound W
  * [1849	Fri Jun 23 09:37:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20230623-093708 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687527428 905 49 Southbound W
  * [1855	Thu Jun 22 12:28:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20230622-122853 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687451333 995 63 Southbound W
  * [1866	Thu Jun 22 21:33:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20230622-213335 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687484015 986 81 Northbound W
  * [1862	Sun Jun 25 23:15:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20230625-231509 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687749309 946 77 Northbound E
  * [1850	Sat Jun 24 09:11:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20230624-091138 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687612298 955 87 Southbound W
  * [1859	Sat Jun 24 12:04:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20230624-120427 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687622667 977 76 Southbound E
  * [1856	Thu Jun 22 23:51:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20230622-235155 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687492315 976 47 Northbound W
---------------------------------------------
Satellite SDR Settings:
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: 0
  noaa_15_enable_bias_tee: false
  noaa_15_gain: 44.5
  noaa_15_sun_min_elevation: 0
  noaa_15_sat_min_elevation: 45
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: 0
  noaa_18_enable_bias_tee: false
  noaa_18_gain: 44.5
  noaa_18_sun_min_elevation: 0
  noaa_18_sat_min_elevation: 45
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: 0
  noaa_19_enable_bias_tee: false
  noaa_19_gain: 44.5
  noaa_19_sun_min_elevation: 0
  noaa_19_sat_min_elevation: 45
  meteor_m2_schedule: false
  meteor_m2_sdr_device_id: 0
  meteor_m2_freq_offset: 0
  meteor_m2_enable_bias_tee: false
  meteor_m2_gain: 44.5
[... 70 lines cut ...]
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));
.

**Server information (please complete the following information):**
 - OS: Raspbian
 - Python version e.g. 3.9.2
 - SDR device:  RTL-SDR v3

**Additional context**
Using fresh image of raspberry-noaa-v2.  I did not change any frequency information in any file/script.

dandspach, 2023-07-11:
No replies, but seems to be working nonetheless.

mihajlo2003petkovic, 2023-10-04:
Spectrogram displays the drequency of the demodulated audio which is around 1-3 kHz, not the baseband file.

closed 2023-10-04 by mihajlo2003petkovic