jekhokie/raspberry-noaa-v2 #588: [Error] Map tiler wrong api key

mistrjirka, 2023-06-26:
**Describe the bug**
Globe is missing in the web UI satellites are orbiting empty space. The problem is that the map is loading with 403 error. 

**Screenshots and/or logs**
[image]
[image]
**Support Output**
=============================================
Details about environment
=============================================
Current local date/time: Mon 26 Jun 21:20:53 CEST 2023
Current date/time (ms):  1687807253
Repo git hash:           645bb9e0a9d15f18bb77a082c5928abc62979aff
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.1
Architecture:            Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [2	Mon Jun 26 21:37:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20230626-213720 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687808240 944 38 Northbound W
  * [1	Mon Jun 26 22:14:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20230626-221418 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1687810458 947 58 Northbound E
---------------------------------------------
Satellite SDR Settings:
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: 0
  noaa_15_enable_bias_tee: true
  noaa_15_gain: 30
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 30
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: 0
  noaa_18_enable_bias_tee: true
  noaa_18_gain: 30
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 30
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: 0
  noaa_19_enable_bias_tee: true
  noaa_19_gain: 30
  noaa_19_sun_min_elevation: 6
  noaa_19_sat_min_elevation: 30
  meteor_m2_schedule: true
  meteor_m2_sdr_device_id: 0
  meteor_m2_freq_offset: 0
  meteor_m2_enable_bias_tee: true
  meteor_m2_gain: 30
  meteor_m2_sun_min_elevation: 6
  meteor_m2_sat_min_elevation: 30
  meteor_m2_memory_threshold: 600
---------------------------------------------
USB Device Map:
  Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
  Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
  Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
  mmcblk0     179:0    0   58G  0 disk 
  ├─mmcblk0p1 179:1    0  256M  0 part /boot
  └─mmcblk0p2 179:2    0 57.7G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks    Used Available Use% Mounted on
  /dev/root       59478240 5164168  51840680  10% /
  devtmpfs          682576       0    682576   0% /dev
  tmpfs             946800       0    946800   0% /dev/shm
  tmpfs             378720    1216    377504   1% /run
  tmpfs               5120       4      5116   1% /run/lock
  /dev/mmcblk0p1    261108   51630    209478  20% /boot
  tmpfs             189360      24    189336   1% /run/user/1000
  tmpfs            1024000       0   1024000   0% /var/ramfs
---------------------------------------------
Memory Info:
  MemTotal:        1893600 kB
  MemFree:          180884 kB
  MemAvailable:    1560184 kB
---------------------------------------------
Git source files changed:
  (None)
---------------------------------------------
SDR Information:
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

gjmkoper, 2023-07-25:
Dear all,

same issue here. I have a key. I would like to know where to put it.

Thanks in advance!

mihajlo2003petkovic, 2023-10-04:
I've fixed it in the latest update. Follow the upgrade guide on Github.

closed 2023-10-04 by mihajlo2003petkovic