jekhokie/raspberry-noaa-v2 #591: [Error] - Globe picture not showing

djvdberg, 2023-07-04:
**Describe the bug**
Globe picture not showing

**Screenshots and/or logs**
[image]

Please help, what am I missing?

Thanks
Dion

ysamouhos, 2023-07-06:
Hi Dion, 
Seems like something wrong with the Maptiler Terrain API. 
You can go to settings and select Terrain = None  (Click the Eye icon on top right, then press the Map Icon and then select none on the Terrain.

djvdberg, 2023-07-10:
Hi,

Works thanks, but every time I refresh I have to go disable it, can that be disabled somewhere in a config?

Thanks
D

ysamouhos, 2023-07-12:
You can edit /var/www/wx-new/App/Views/Passes/index.html. 
line 11 add &terrain=None in the url. 

example:  ...src="https://satvis.space/?elements=Point,Label,SensorCon,Orbit&layers=ArcGis**&terrain=None**&gs=....
.

gjmkoper, 2023-08-06:
Works great, thanks.

mihajlo2003petkovic, 2023-10-04:
This was fixed. Maptiler was changed to ArcGis. Update setups.

closed 2023-10-04 by mihajlo2003petkovic