jekhokie/raspberry-noaa-v2 #600: AUDIO OUTPUT

EASDK8600, 2023-07-21:
**Please Describe the problem to be solved**
During debugging it would be very nice to have an audiosignal when satellits are passing

**(Optional): Suggest A Solution**
The Raspberry Pi's (A pi3b and above) has a audio output (3.5 mm Jack) and I was just thinking of a option switch in the CONFIG file to enable / disable audio output from the passing satellits.

Best regards 
Esben, Denmark

bradjtrammell, 2023-08-06:
+1 for this. I think it be great for debugging! I have to use a Windows laptop with an SDR++ setup split off my NOAA setup to do live monitoring and it's a PITA. It'd be great to do it natively and just have live monitoring be an option.

EASDK8600, 2023-08-06:
> +1 for this. I think it be great for debugging! I have to use a Windows laptop with an SDR++ setup split off my NOAA setup to do live monitoring and it's a PITA. It'd be great to do it natively and just have live monitoring be an option.

Thanks Brad.... Exsagtly My thoughts.
My Project will have an ampilifer and speakers build in the Box anyways... So audio signal of satellits passing would just be great during debugging (or just for fun 🤭)
/Esben., Denmark

scm2000, 2023-08-25:
For similar reasons, debugging, what I did was install icecast2 and ezstream, then I added a tee command on the noaa script for receiving, so basically rtl_fm pipes to tee, and tee pipes to both the original sox command to write the file, and another sox command piping to ezstream...
Now I am able to go to the icecast mount point url with chrome on my laptop and listen to the live audio stream. The stream automatically starts at the begining of the pass and shuts down at the end.
This works better for  me than getting audio out the audio jack,  since ultimately my raspbery pi is going to be inside a waterproof box in my garden, run by a battery and solar panel-- the begining of my garden weather station.

mihajlo2003petkovic, 2023-10-04:
This could technically be implemented for both GNU Radio and RTL_FM, but not for SatDump live reception. Although isn't it better to just use GQRX for live reception? That's an SDR app and is used to receive signals.