jekhokie/raspberry-noaa-v2 #618: Please expand to support amd64 Ubuntu

bradjtrammell, 2023-07-28:
I would absolutely love to see this expanded to work natively with AMD64 architecture chipsets as well, especially with the skyrocketing costs of raspberry pi computers. Most hobbyists have fairly consistent access to thin clients and without modification of the ansible tasks, this can't be run natively to install this core set of scripts and programs.

Mostly this was because of WXtoImg and the having specific packages hard coded in for ARM, which were easily available via Apt on Ubuntu. I assume this must not be the case on Raspbian. It would be ideal to add a if/else/then switch to detect if the arch is x86_64 or arm, and then force the set of ansible tasks based on that removing the need for it to force install an ARM package if the user is using an x86_64 arch.

With those arch specific tasks removed. This installed perfectly on amd64, but it doesn't appear to be saving my configuration changes. That's probably due to my own incompetence though.

Cadair, 2023-07-28:
See #461 for some effort along these lines.

mihajlo2003petkovic, 2023-10-04:
Fixed.

mihajlo2003petkovic, 2023-10-04:
Now RN2 natively works with amd64.

closed 2023-10-04 by mihajlo2003petkovic