jekhokie/raspberry-noaa-v2 #620: database error

pb2888, 2023-07-29:
Fresh install,done exactly what it says to do and when I go to access the webpage I get this error

Fatal error: Uncaught Exception: Unable to open database: unable to open database file in /var/www/wx-new/Lib/Controller.php:46 Stack trace: #0 /var/www/wx-new/Lib/Controller.php(46): SQLite3->__construct() #1 /var/www/wx-new/Lib/Controller.php(12): Lib\Controller->connectToDB() #2 /var/www/wx-new/Lib/Router.php(50): Lib\Controller->__construct() #3 /var/www/wx-new/Lib/Router.php(13): Lib\Router->handleRequest() #4 /var/www/wx-new/public/index.php(16): Lib\Router->__construct() #5 {main} thrown in /var/www/wx-new/Lib/Controller.php on line 46

closed 2023-08-02 by pb2888

bradjtrammell, 2023-10-05:
@pb2888, If you closed the issue, you must have resolved your issue...

Do you mind sharing it so others can benefit from it as well in case they come across the same issue?

jtadesse977, 2024-04-21:
Please share if you solved this because I am getting same issue and I've tried so much