jekhokie/raspberry-noaa-v2 #633: [Error] Capture is working but system won't decode

bradjtrammell, 2023-08-06:
**Describe the bug**
So I've had two systems running side by side here for a few weeks, this one which is supposed to be automatic (rPi NOAA v2) and one I've been doing manually with each pass. Both using the same antenna and the one I run manually on each pass using and older laptop running Windows 10, SDR++ and wxToImg captures and works beautifully. 

However on that exact same pass, my linux install running the exact same model of SDR as on the laptop (mind you running through the same filters and antennas as my windows one that works fine and captures with no issues) , either doesn't spit out an image, or just randomly stops recording halfway through the pass with no error message or citing that it can't detect a signal (or something similar) even when while monitoring the pass on Windows I can clearly hear a clean signal. Just for giggles, I swapped the SDR's between passes (thinking maybe the linux SDR was bad) and it captured with no issues on the Windows machine and the known good working SDR that was working on Windows now was not functioning on the linux machine.

I am at my wits end and have tried both "gnuradio" and "rtl_fm", and I've changed every single setting in settings.yml at least 100 times trying to see if maybe it wasn't being dialed in properly.

**Screenshots and/or logs**
This is the last pass that was attempted, the Windows computer captured a beautiful image from this with no issues, whereas the Ubuntu machine spat out this:

INFO : Recording at 137.9125 MHz...
06-08-2023 13:14 /home/pi/raspberry-noaa-v2/scripts/audio_processors/noaa_record_rtl_fm.sh INFO : Recording at 137.9125 MHz...
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R828D tuner
Tuner gain set to 49.60 dB.
activated bias-T on GPIO PIN 0
Tuner error set to 20 ppm.
Tuned to 138392500 Hz.
Oversampling input by: 32x.
Oversampling output by: 1x.
Buffer size: 4.27ms
Sampling at 1920000 S/s.
Output at 60000 Hz.
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
cb transfer status: 5, canceling...
Signal caught, exiting!

User cancel, exiting...
06-08-2023 13:28 /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Producing spectrogram
ffmpeg version 4.2.7-0ubuntu0.1 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-nvenc --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
Guessed Channel Layout for Input Stream #0.0 : mono
Input #0, wav, from '/srv/audio/noaa/NOAA-18-20230806-131401.wav':
  Duration: 00:05:19.49, bitrate: 176 kb/s
    Stream #0:0: Audio: pcm_s16le ([1][0][0][0] / 0x0001), 11025 Hz, mono, s16, 176 kb/s
Stream mapping:
  Stream #0:0 (pcm_s16le) -> showspectrumpic
  showspectrumpic -> Stream #0:0 (png)
Press [q] to stop, [?] for help
Output #0, image2, to '/home/pi/raspberry-noaa-v2/tmp/temp_spectrogram.png':
  Metadata:
    encoder         : Lavf58.29.100
    Stream #0:0: Video: png, rgb24, 652x928 [SAR 1:1 DAR 163:232], q=2-31, 200 kb/s, 10.77 fps, 10.77 tbn, 10.77 tbc
    Metadata:
      encoder         : Lavc58.54.100 png
[... 203 lines cut ...]
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

**Server information (please complete the following information):**
 - OS: [e.g. Raspbian]
 - Python version [e.g. 3.7.3]
 - SDR device [e.g. RTL-SDR v3]

**Additional context**
Just to clear things up, to simplify the install I created a default user on my linux machine called Pi, as most of the scripts depend on that user or directory existing to run. I also installed all of the scripts that are defined for ARM manually. 

Additionally, this worked at one point for about 3 passes before ultimately never capturing again. I also wanted to mention (because I'm sure someone will bring it up), yes I have the antenna split going into each machine, but I'm also using the NOAA Sawbird Filter, an FM Filter and an Coaxial Amplifier to boost the signal. Which clearly shouldn't be the issue as the Windows machine is recieving data beautifully with pristine quality images.

pb2888, 2023-08-11:
looks to me ur RTL drivers arent blacklisted,make sure ya got the proper GPS location,update the keplers etc but if it doesnt work start again and use the raspberry noaa image if ya didnt already,btw if u set the gain to 0.0 it makes it auto which WILL help as ya said ur usin a LNA which will give ya front end overload which will kill the signal and will damage the receive on the dongle

bradjtrammell, 2023-08-23:
Hi there. Thanks for your reply. I was beginning to lose hope that I would even get a reply :-(.

This isn't a Raspberry Pi setup, its an Ubuntu Server install on an old laptop (as noted in my original post), that meets all the same requirements as the Raspberry Pi version (they are so insanely expensive it is cheaper to run this on an old laptop).

Either way, the software appears to be "working" and decoding now all of a sudden, but now I'm barely getting a usable signal despite having an unobstructed view on certain passes. I have tinkered with it, removed and added filters, LNA's, amplifiers, etc., and this right here is the best it's ever been. 

[image]
[image]

Even on passes where a satellite is directly overhead on a clear, blue skyed day, I'm still getting this level of signal. It's extremely frustrating, because the Windows PC is getting a clean and clear signal. I'm also noticing that it starts recording almost 5 minutes too early, and keeps recording almost 5 minutes after the satellite should have already passed. 

Am I supposed to have a clear line of sight for the entire 15 minutes of the pass, because my understanding was that as long as enough telemetry was received to decode for 3 mins or so, you were essentially getting the data needed to get an image. Is this not the case?

mihajlo2003petkovic, 2023-10-04:
The lengths of the image purely depends on the signal strength during the whole pass. Whenever signal drops, you get noise.

bradjtrammell, 2024-01-01:
> @bradjtrammell I've updated the system so it works with Linux PCs natively. Try updating your RN2 version by deleting settings.ymk file, running git pull, filling out your newly created settings.yml file and running the install script.

Apologies for the massive delay. I see the changes you've mentioned and have been using for several months not even realizing these changes were made because of my post. So thank you so much for that. 

I recently upgraded my server to Ubuntu 22 not realizing it would break the functionality so I now have to start all over and reimage my system but it gives me the chance to go through the setup again :-)

closed 2024-12-02 by MihajloPi