jekhokie/raspberry-noaa-v2 #708: Meteor2

f5pql, 2023-09-12:
Hello,

With a fresh install, I decode without problem NOAA 18 & 19 but for meteor nothing :( in the wav only silent , there is no noise, nothing.

Thanks for your help

12-09-2023 18:04 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : The system have enough space to store a Meteor pass on RAM
12-09-2023 18:04 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Free : 761 ; Required : 600
12-09-2023 18:04 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Direction Northbound
12-09-2023 18:04 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : I'll flip this image pass because FLIP_METEOR_IMG is set to true and PASS_DIRECTION is Northbound
12-09-2023 18:04 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Starting rtl_fm record
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (fr_FR)
INFO : Recording at 137.1000 MHz...
12-09-2023 18:04 /home/pi/raspberry-noaa-v2/scripts/audio_processors/meteor_record_rtl_fm.sh INFO : Recording at 137.1000 MHz...
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0013 tuner
Tuner gain set to 19.70 dB.
Tuned to 137388000 Hz.
Oversampling input by: 4x.
Oversampling output by: 1x.
Buffer size: 7.11ms
Sampling at 1152000 S/s.
Output at 288000 Hz.
Allocating 15 zero-copy buffers
Signal caught, exiting!

User cancel, exiting...
12-09-2023 18:19 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Demodulation in progress (QPSK)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (fr_FR)
Input: /var/ramfs/METEOR-M-2-20230912-180445.wav, output: /home/pi/raspberry-noaa-v2/tmp/meteor/METEOR-M-2-20230912-180445.qpsk
Demodulator initialized

(  0.0%) Carrier:    +0.0 Hz, Symbol rate: 72000.0 Hz, Locked: No
(  1.6%) Carrier: -3108.8 Hz, Symbol rate: 72001.5 Hz, Locked: No
(  3.1%) Carrier: -1886.3 Hz, Symbol rate: 72003.9 Hz, Locked: No
(  4.7%) Carrier:  +102.2 Hz, Symbol rate: 71998.5 Hz, Locked: No
(  6.3%) Carrier: +2482.3 Hz, Symbol rate: 72001.7 Hz, Locked: No
(  7.8%) Carrier: +2024.4 Hz, Symbol rate: 72005.8 Hz, Locked: No
(  9.4%) Carrier:   -39.7 Hz, Symbol rate: 72003.8 Hz, Locked: No
( 10.9%) Carrier: -1788.6 Hz, Symbol rate: 72008.6 Hz, Locked: No
( 12.5%) Carrier: -2845.9 Hz, Symbol rate: 72011.1 Hz, Locked: No
( 14.1%) Carrier:  -612.8 Hz, Symbol rate: 72010.1 Hz, Locked: No
( 15.6%) Carrier: +1662.1 Hz, Symbol rate: 72009.1 Hz, Locked: No
( 17.2%) Carrier: +3150.5 Hz, Symbol rate: 72006.8 Hz, Locked: No
( 18.7%) Carrier:  +154.3 Hz, Symbol rate: 72007.5 Hz, Locked: No
( 20.3%) Carrier: -1924.2 Hz, Symbol rate: 72002.2 Hz, Locked: No
( 21.9%) Carrier: -2492.3 Hz, Symbol rate: 71997.1 Hz, Locked: No
( 23.4%) Carrier:  +206.7 Hz, Symbol rate: 71996.5 Hz, Locked: No
( 25.0%) Carrier: +1922.4 Hz, Symbol rate: 71999.0 Hz, Locked: Yes
( 26.5%) Carrier: +1461.5 Hz, Symbol rate: 71999.4 Hz, Locked: No
( 28.1%) Carrier:  -429.6 Hz, Symbol rate: 72002.8 Hz, Locked: No
( 29.6%) Carrier: -2233.5 Hz, Symbol rate: 72004.5 Hz, Locked: No
( 31.2%) Carrier: -2929.5 Hz, Symbol rate: 72012.8 Hz, Locked: No
( 32.8%) Carrier: -1045.1 Hz, Symbol rate: 72012.1 Hz, Locked: No
( 34.3%) Carrier:  +584.9 Hz, Symbol rate: 72017.6 Hz, Locked: No
( 35.9%) Carrier: +2978.2 Hz, Symbol rate: 72011.0 Hz, Locked: No
( 37.5%) Carrier: +1277.9 Hz, Symbol rate: 72009.2 Hz, Locked: No
( 39.1%) Carrier: -1847.0 Hz, Symbol rate: 72014.9 Hz, Locked: No
( 40.6%) Carrier: -3053.9 Hz, Symbol rate: 72016.1 Hz, Locked: No
( 42.2%) Carrier: -1409.3 Hz, Symbol rate: 72016.9 Hz, Locked: No
( 43.8%) Carrier:  +176.1 Hz, Symbol rate: 72008.3 Hz, Locked: No
( 45.4%) Carrier: +2006.3 Hz, Symbol rate: 72008.2 Hz, Locked: No
( 46.9%) Carrier: +2783.2 Hz, Symbol rate: 72013.8 Hz, Locked: No
( 48.5%) Carrier:  +677.4 Hz, Symbol rate: 72017.2 Hz, Locked: No
( 49.1%) Carrier: +2372.6 Hz, Symbol rate: 72013.4 Hz, Locked: No
( 50.6%) Carrier: +1962.3 Hz, Symbol rate: 72000.1 Hz, Locked: No
( 52.2%) Carrier:  -862.2 Hz, Symbol rate: 72001.1 Hz, Locked: Yes
[... 174 lines cut ...]
12-09-2023 18:38 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Producing polar graph of direction for pass
convert: unable to open image `spread_*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
convert: no images defined `spread_*.jpg' @ error/convert.c/ConvertImageCommand/3229.
12-09-2023 18:38 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Annotating images and creating thumbnails
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (fr_FR)
Loading page (1/2)
[>                                                           ] 0%
[======>                                                     ] 10%
[=========>                                                  ] 15%
[=============>                                              ] 23%
[============================================================] 100%
Rendering (2/2)                                                    
[>                                                           ] 0%
[===============>                                            ] 25%
[============================================================] 100%
Done                                                               
convert: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
convert: image sequence is required `-composite' @ error/mogrify.c/MogrifyImageList/8003.
convert: no images defined `*.jpg' @ error/convert.c/ConvertImageCommand/3229.
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (fr_FR)
convert: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
convert: no images defined `*-thumb-122-rectified.jpg' @ error/convert.c/ConvertImageCommand/3229.
12-09-2023 18:38 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh ERROR : Did not get a successful .bmp image - stopping processing
12-09-2023 18:38 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Total processing time: 00:34.55
Could not open input file

scm2000, 2023-09-12:
If you have pulled code from the main branch, then it's trying to receive Meteor M2.  Meteor M2 is not transmitting LRPT, maybe nothing at all actually, it's broken.    Meteor M2-3 has come online and is working, and they are working on supporting it in the beta-development branch.

f5pql, 2023-09-13:
Ok, thanks for this info ... To test the branch beta, have you the address git for this ?

f5pql, 2023-09-15:
With a new fresh install, I decode without problem NOAA 18 & 19 but for meteor nothing :( in the wav only silent , there is no noise, nothing for [METEOR-M2 3]

 Carrier: -357.64Hz	 Lock detector: 0.23	 isLocked: 0	 OutputSize: 125.32Mb Progress: 99.98% 		
 Carrier: -359.00Hz	 Lock detector: 0.23	 isLocked: 0	 OutputSize: 125.33Mb Progress: 99.99% 		
 Carrier: -360.14Hz	 Lock detector: 0.23	 isLocked: 0	 OutputSize: 125.34Mb Progress: 100.00% 		
SyncWordFound:1 | Decoded Packets:0 | Current Pos:11851436 | Phase:1 | synch:912b3328 | RS: (-1, -1, -1, -1)		
SyncWordFound:2 | Decoded Packets:0 | Current Pos:14187304 | Phase:6 | synch:a99f5c90 | RS: (-1, -1, -1, -1)		
SyncWordFound:3 | Decoded Packets:0 | Current Pos:16010125 | Phase:10 | synch:69694843 | RS: (-1, -1, -1, -1)		
SyncWordFound:4 | Decoded Packets:0 | Current Pos:16075304 | Phase:5 | synch:dc5b3a6f | RS: (-1, -1, -1, -1)		
SyncWordFound:5 | Decoded Packets:0 | Current Pos:19818319 | Phase:6 | synch:d8cf6772 | RS: (-1, -1, -1, -1)		
SyncWordFound:6 | Decoded Packets:0 | Current Pos:19847137 | Phase:9 | synch:186982d5 | RS: (-1, -1, -1, -1)		
SyncWordFound:7 | Decoded Packets:0 | Current Pos:20579444 | Phase:0 | synch:f87b51db | RS: (-1, -1, -1, -1)		
SyncWordFound:8 | Decoded Packets:0 | Current Pos:21672634 | Phase:0 | synch:2b5186a0 | RS: (-1, -1, -1, -1)		
SyncWordFound:9 | Decoded Packets:0 | Current Pos:35142491 | Phase:6 | synch:ed6f07e | RS: (-1, -1, -1, -1)		
SyncWordFound:10 | Decoded Packets:0 | Current Pos:50647238 | Phase:0 | synch:faae5ec6 | RS: (-1, -1, -1, -1)		
SyncWordFound:11 | Decoded Packets:0 | Current Pos:52591160 | Phase:5 | synch:88a7b17d | RS: (-1, -1, -1, -1)		
SyncWordFound:12 | Decoded Packets:0 | Current Pos:57227135 | Phase:14 | synch:beae66f9 | RS: (-1, -1, -1, -1)		
SyncWordFound:13 | Decoded Packets:0 | Current Pos:58499710 | Phase:0 | synch:37432f88 | RS: (-1, -1, -1, -1)		
SyncWordFound:14 | Decoded Packets:0 | Current Pos:60963395 | Phase:9 | synch:b98e5277 | RS: (-1, -1, -1, -1)		
SyncWordFound:15 | Decoded Packets:0 | Current Pos:81549925 | Phase:7 | synch:7682a37d | RS: (-1, -1, -1, -1)		
SyncWordFound:16 | Decoded Packets:0 | Current Pos:121631862 | Phase:4 | synch:dbcf855f | RS: (-1, -1, -1, -1)		
SyncWordFound:17 | Decoded Packets:0 | Current Pos:130961760 | Phase:4 | synch:5c8266d7 | RS: (-1, -1, -1, -1)		
No data received, try to make composite images
Generate composite images
Generate composite images done
convert: unable to open image `spread_*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
convert: no images defined `spread_*.jpg' @ error/convert.c/ConvertImageCommand/3229.
Loading page (1/2)
[>                                                           ] 0%
[======>                                                     ] 10%
[=========>                                                  ] 15%
[=============>                                              ] 23%
[============================================================] 100%
Rendering (2/2)                                                    
[>                                                           ] 0%
[===============>                                            ] 25%
[============================================================] 100%
Done                                                               
convert: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
convert: image sequence is required `-composite' @ error/mogrify.c/MogrifyImageList/8003.
convert: no images defined `/srv/images/METEOR-M2-3-20230915-112606-*.jpg' @ error/convert.c/ConvertImageCommand/3229.
convert: unable to open image `/srv/images/METEOR-M2-3-20230915-112606-*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
convert: no images defined `/srv/images/thumb/METEOR-M2-3-20230915-112606-*.jpg' @ error/convert.c/ConvertImageCommand/3229.
15-09-2023 12:14 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Deleting audio files
15-09-2023 12:14 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : No images found, not pushing anything
15-09-2023 12:14 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Total processing time: 00:48.30

jp112sdl, 2023-09-21:
METEOR-M2-3 decoding works for me, using the develop-branch and `satdump_live`.
With `gnuradio` I had no success at all.

[image]

f5pql, 2023-09-22:
Hi,
Thanks for this, it's possible to send me the frequency that you use 137.1 or 137.9 ? have you the link of the develop-branch. For me actually used the .img of https://qsl.net/ve3elb/RaspiNOAA/ ond rtl_sdl .... I'll test gnuradio, perhasp it's better ...

jp112sdl, 2023-09-22:
> have you the link of the develop-branch. 

[image]

>  the frequency that you use 

https://github.com/jekhokie/raspberry-noaa-v2/blob/09c98e0bbf1af6dfae84ca78d9a78c7907545ed4/scripts/common.sh#L55

f5pql, 2023-09-22:
Many thanks for all informations.
It's possible to send me your config/settings.yml to compare with mine to find where is my mistake ....

f5pql, 2023-09-22:
> > have you the link of the develop-branch.
> 
> [image]
> > the frequency that you use
> 
> https://github.com/jekhokie/raspberry-noaa-v2/blob/09c98e0bbf1af6dfae84ca78d9a78c7907545ed4/scripts/common.sh#L55

ok, I'm test a new install of this branch, meteor pass at 12h05 local time .... I hope a good result ;)

f5pql, 2023-09-23:
Hi,
I wanted to have a config file that works correctly for someone and adapt it if necessary for my needs. For the moment I receive the NOAA VERY WELL but for Meteor the wav is dead silence, I don't understand why, that's why I asked for help...

mihajlo2003petkovic, 2023-09-23:
Can you share a sample NOAA image?
Meteor is a bit harder to receive, has lower signal strength. Try to receive Meteor manually using your laptop and finding the best gain then set your Meteor gain to that value.

f5pql, 2023-09-23:
For the moment, following the advice of jp112sdl, I have installed the development branch and I am in the process of setting it up correctly. An image of NOAA19 is in progress, I think I can send it to you as soon as it is finished processing. For Meteor I have dead silence, not even noise, that's why I'm asking questions, and above all I don't yet have all the knowledge to solve this problem.

f5pql, 2023-09-23:
During processing, I just got this image...

[image]

mihajlo2003petkovic, 2023-09-23:
And what gain did you use for Meteor? The same one as for NOAA?

f5pql, 2023-09-23:
for the moment, yes

f5pql, 2023-09-23:
but, it's a fresh install from two hours ;)

jp112sdl, 2023-09-23:
My settings:
```
METEOR_M2_3_GAIN=25.4
NOAA_xx_GAIN=19.7

RECEIVER_TYPE=rtlsdr
NOAA_RECEIVER=rtl_fm
METEOR_RECEIVER=satdump_live
```

using a Nooelec RTL-SDR v4 SDR with a general purpose +20dB LNA (8-2500MHz) and this DIY antenna:
[image]

f5pql, 2023-09-24:
For the last meteor passages it is always a failure. Perhaps not enough gain which would explain the impossibility of decoding. We will continue the investigations and try to improve and recover db in reception....

jp112sdl, 2023-09-25:
> Perhaps not enough gain which would explain the impossibility of decoding.

For me it only works within a small range. A gain below 20 is too low and above 29 is too much.

mihajlo2003petkovic, 2023-10-03:
Did you get your Meteor reception working?

f5pql, 2023-10-04:
No, NOAA works very well but Meteor nothing. I choise the pass with max elevation, and I've test with rtl_fm and satdump_live.
If you are a ideas ;)

f5pql, 2023-10-05:
It's normal that noaa file wav have ~12Mo with sound  and  meteor wav have 350Mo and this a silent with no sound or noise .?

mihajlo2003petkovic, 2023-10-05:
Yes, it is. Meteor is digital signal and will sound like noise.

f5pql, 2023-10-06:
Oki, but the meteor wav have nothing very nothing only a pure silent ;)

f5pql, 2023-10-08:
Yeah !!! After many many many parameters and log analyse MY 1er meteor jpg !!!! nothing modify with hard configuration, only I choise to use gnuradio to create the wav.
Many thanks all that help me to resolved this ....
Just a last question, it's possible to impose a Aos and Los to jpg image and cut the beginning et last of image to purge noise with minimal degree..

[image]

closed 2024-04-12 by MihajloPi