jekhokie/raspberry-noaa-v2 #722: Variable is not replaced with username

jp112sdl, 2023-09-21:
When the logrotate settings are deployed, the variable `{{ target_user }}` ist not replaced with the username.

https://github.com/jekhokie/raspberry-noaa-v2/blob/09c98e0bbf1af6dfae84ca78d9a78c7907545ed4/ansible/roles/common/files/logrotate-settings#L7
https://github.com/jekhokie/raspberry-noaa-v2/blob/09c98e0bbf1af6dfae84ca78d9a78c7907545ed4/ansible/roles/common/files/email-logrotate-settings#L7

This leads to a logrotate service error.
```
systemd[1]: Starting Rotate log files...
logrotate[8446]: error: msmtp:7 extra arguments for create
logrotate[8446]: error: found error in /var/log/msmtp/*.log , skipping
logrotate[8446]: error: found error in file msmtp, skipping
logrotate[8446]: error: raspberry-noaa-v2:7 extra arguments for create
logrotate[8446]: error: found error in /var/log/raspberry-noaa-v2/*.log , skipping
logrotate[8446]: error: found error in file raspberry-noaa-v2, skipping
systemd[1]: logrotate.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: logrotate.service: Failed with result 'exit-code'.
systemd[1]: Failed to start Rotate log files.

```

_As a workaround, I edited the files `/etc/logrotate.d/raspberry-noaa-v2` and `/etc/logrotate.d/msmtp` manually._

mihajlo2003petkovic, 2023-09-22:
I'll look into this.

mihajlo2003petkovic, 2023-10-03:
Fixed.

closed 2023-10-03 by mihajlo2003petkovic