jekhokie/raspberry-noaa-v2 #763: 'predict': buffer overflow, if path (including filename) to TLE file is longer than 48 characters

jp112sdl, 2023-10-03:
If RN2 is installed within another user environment (not 'pi') and the username exceeds 11 characters, the path to the TLE file (`/home/<username>/raspberry-noaa-v2/tmp/orbit.tle`) gets too long and leads to a buffer overflow in `predict`.

> buffer overflow detected : terminated
Aborted

A fix must be implemented in `predict`:

```C++
		if (strcmp(argv[x],"-t")==0)
		{
			z=x+1;
			if (z<=y && argv[z][0] && argv[z][0]!='-')
				strncpy(tle_cli,argv[z],48);
		}
```
https://github.com/kd2bd/predict/blob/517fccf421909b4974151b6ca30810586ca658a3/predict.c#L6303

mihajlo2003petkovic, 2023-10-03:
Good catch!

jp112sdl, 2023-10-04:
Maybe a fast fix could be doing a `cd ${NOAA_HOME}/tmp` and using `predict -t ./orbit.tle` (without the full path)

albertfilice, 2024-03-26:
Hitting this, is there a workaround other than moving to a different user?