jekhokie/raspberry-noaa-v2 #769: [Error]sendmail: cannot log to /var/log/msmtp/output.log: cannot open: Permission denied

maserowik, 2023-10-03:
**Describe the bug**
Trying to send email from the schedule.sh 
Spoke with **Mihaijlo P**i and asked to open a issue on github.

**Screenshots and/or logs**
(sendmail: cannot log to /var/log/msmtp/output.log: cannot open: Permission denied)

**Support Output**
noaa@hp-noaa:~/raspberry-noaa-v2$ ./support.sh
cat: /proc/device-tree/model: No such file or directory
=============================================
Details about environment
=============================================
Current local date/time: Tue 03 Oct 2023 05:41:19 PM EDT
Current date/time (ms):  1696369279
Repo git hash:           afddfac1ebe4cd7a28be1cad9c4c4a1a8e8dafb5
Raspberry Pi Model:
Architecture:            Linux hp-noaa 5.10.0-25-amd64 #1 SMP Debian 5.10.191-1 (2023-08-16) x86_64 GNU/Linux
Num CPUs:                2
node0
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [1027       Sat Oct  7 10:24:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231007-102444 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696688684 954 57 Southbound E
  * [1003       Fri Oct  6 23:55:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231006-235514 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696650914 926 40 Northbound W
  * [993        Sun Oct  8 08:54:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231008-085429 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696769669 944 66 Southbound E
  * [995        Mon Oct  9 08:29:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231009-082904 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696854544 895 38 Southbound E
  * [1002       Fri Oct  6 12:29:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231006-122916 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696609756 953 69 Southbound W
  * [1007       Sun Oct  8 23:30:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231008-233019 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696822219 955 67 Northbound W
  * [1017       Fri Oct  6 21:55:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231006-215540 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696643740 977 54 Northbound W
  * [1000       Thu Oct  5 12:41:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231005-124138 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696524098 964 55 Southbound W
  * [987        Thu Oct  5 08:31:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231005-083125 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696509085 919 40 Southbound E
  * [998        Wed Oct  4 12:54:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231004-125401 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696438441 911 42 Southbound W
  * [1026       Fri Oct  6 10:47:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231006-104743 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696603663 960 88 Southbound W
  * [1009       Mon Oct  9 23:17:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231009-231759 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696907879 996 87 Northbound W
  * [986        Wed Oct  4 20:16:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231004-201611 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696464971 927 79 Northbound W
  * [988        Thu Oct  5 19:50:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231005-195046 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696549846 946 60 Northbound E
  * [1005       Sat Oct  7 23:42:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231007-234244 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696736564 971 52 Northbound W
  * [1012       Wed Oct  4 20:39:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231004-203950 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696466390 941 32 Northbound E
  * [1021       Sun Oct  8 21:31:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231008-213117 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696815077 959 89 Northbound W
  * [1029       Mon Oct  9 11:18:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231009-111841 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696864721 930 46 Southbound W
  * [1023       Mon Oct  9 21:19:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231009-211912 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696900752 955 71 Northbound E
  * [1001       Thu Oct  5 22:27:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231005-222713 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696559233 899 33 Northbound E
  * [1028       Sun Oct  8 10:01:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231008-100151 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696773711 932 34 Southbound E
  * [992        Sat Oct  7 20:39:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231007-203931 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696725571 927 48 Northbound W
  * [1004       Sat Oct  7 12:16:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231007-121655 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696695415 998 89 Southbound W
  * [1014       Thu Oct  5 10:41:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231005-104159 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696516919 1007 74 Southbound W
  * [989        Fri Oct  6 09:45:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231006-094541 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696599941 927 42 Southbound W
  * [984        Tue Oct  3 20:41:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231003-204155 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696380115 939 46 Northbound W
  * [1011       Wed Oct  4 10:54:00 2023 a noaa] -> /home/noaa/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231004-105410 /home/noaa/raspberry-noaa-v2/tmp/orbit.tle                                                               1696431250 948 59 Southbound W
[... cut ...]
**Additional context**
It appears that Apparmor will not allow loggin to the /var/log/msmtp/output.log.
AppArmor website is
https://apparmor.net/

My work around was the following to allow the log file to run was the following
use some of the information from the following website
https://askubuntu.com/questions/878288/msmtp-cannot-write-to-var-log-msmtp-msmtp-log

**my steps were as follows.
cd /etc/apparmor.d
sudo nano usr.bin.msmtp
add  
owner /var/log/msmtp/output.log wk,

service apparmor restart**

mihajlo2003petkovic, 2023-10-03:
I'll investigate!

maserowik, 2023-10-04:
Oct 04 00:01:35 host=smtp.gmail.com tls=on auth=on user=xxx@gmail.com from=xxxx@gmail.com recipients=xxx@verizon.net mailsize=1448902 smtpstatus=250 smtpmsg='250 2.0.0 OK  1696392095 b7-20020a05620a126700b00767d8e12ce3sm97961                                                                                                                     3qkl.49 - gsmtp' exitcode=EX_OK
Oct 04 00:01:35 host=smtp.gmail.com tls=on auth=on user=xxx@gmail.com from=xxx@gmail.com recipients=noaa smtps                                                                                                                     tatus=553 smtpmsg='553-5.1.3 The recipient address <noaa> is not a valid RFC 5321 address. Please\n553-5.1.3 visit\n553-                                                                                                                     5.1.3  https://support.google.com/a/answer/3221692 and review RFC 5321\n553 5.1.3 specifications for more information. w                                                                                                                     25-20020a0ca819000000b0065b2f4dd300sm1030146qva.90 - gsmtp' errormsg='recipient address noaa not accepted by the server' exitcode=EX_DATAERR
This is working with my gmail account.  but now the system is pulling a noaa which is my username for login to the computer.  any idea on how to solve this?

mihajlo2003petkovic, 2023-10-09:
Okay, it appears you solved the issue, but one question: did you install Apparmor? I don't remember anything named like that in RN2. For me stock RN2 mail system worked great.

maserowik, 2023-10-09:
I did not install Apparmor.  I download the lmde from a linux mint site.