jekhokie/raspberry-noaa-v2 #793: After new clean install NO predicted passes on website

jgrolik, 2023-10-06:
**Describe the bug**
I am trying to improve my reception station. Because I installed a newer satdump and I was unable to make ./install_and_upgrade.sh because of version conflict that is why I made a new clean clone of raspberry-noaa-v2 and after ./install_and_upgrade.sh now predict gave 0 passes on the website.

**Screenshots and/or logs**
If applicable, add screenshots to help explain your problem. Logs are always welcome as well.

**Support Output**
=============================================
Details about environment
=============================================
Current local date/time: Fri  6 Oct 19:27:04 CEST 2023
Current date/time (ms):  1696613224
Repo git hash:           8510fbd5b2d309b41efe94264f39070140709bee
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.2
Architecture:            Linux noaa 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [169	Sun Oct  8 22:45:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231008-224512 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696797912 948 64 Northbound W
  * [177	Wed Oct 11 22:08:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231011-220815 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697054895 947 67 Northbound E
  * [191	Thu Oct 12 21:38:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231012-213847 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697139527 956 33 Northbound W
  * [139	Fri Oct  6 23:10:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231006-231007 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696626607 931 41 Northbound W
  * [168	Sun Oct  8 12:55:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231008-125509 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696762509 891 31 Southbound W
  * [197	Thu Oct 12 11:02:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231012-110216 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697101336 906 44 Southbound W
  * [153	Sat Oct  7 09:37:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231007-093727 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696664247 924 41 Southbound E
  * [192	Sun Oct  8 10:54:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231008-105424 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696755264 923 50 Southbound W
  * [156	Sat Oct  7 21:03:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231007-210304 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696705384 925 81 Northbound W
  * [196	Thu Oct 12 09:22:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231012-092218 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697095338 894 31 Southbound E
  * [170	Mon Oct  9 11:01:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231009-110159 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696842119 975 38 Southbound E
  * [165	Thu Oct 12 08:05:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231012-080531 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697090731 937 47 Southbound E
  * [188	Wed Oct 11 20:09:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231011-200951 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697047791 986 55 Northbound E
  * [194	Tue Oct 10 10:08:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231010-100815 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696925295 932 71 Southbound E
  * [185	Tue Oct 10 10:31:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231010-103137 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696926697 977 60 Southbound W
  * [195	Wed Oct 11 09:45:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231011-094515 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697010315 920 48 Southbound E
  * [163	Wed Oct 11 08:31:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231011-083103 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697005863 922 73 Southbound E
  * [146	Sat Oct  7 08:33:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231007-083326 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696660406 944 77 Southbound E
  * [175	Tue Oct 10 22:20:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231010-222031 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696969231 968 83 Northbound E
  * [189	Thu Oct 12 10:07:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231012-100718 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697098038 970 89 Southbound E
  * [144	Fri Oct  6 19:47:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231006-194706 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696614426 861 27 Northbound E
  * [178	Thu Oct 12 12:05:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231012-120539 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697105139 979 70 Southbound W
  * [145	Fri Oct  6 21:26:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20231006-212614 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696620374 926 54 Northbound W
  * [184	Mon Oct  9 20:33:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231009-203357 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696876437 1006 82 Northbound E
  * [166	Thu Oct 12 19:30:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231012-193002 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697131802 915 75 Northbound W
  * [157	Sun Oct  8 08:07:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20231008-080755 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696745275 961 49 Southbound E
  * [176	Wed Oct 11 12:18:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20231011-121801 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1697019481 932 58 Southbound W
  * [182	Sun Oct  8 20:46:00 2023 a pi] -> /home/pi/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20231008-204605 /home/pi/raspberry-noaa-v2/tmp/orbit.tle                                                               1696790765 955 82 Northbound W
[... cut ...]
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

mihajlo2003petkovic, 2023-10-06:
To solve the error, run sudo apt remove --purge satdump and then run the install script to update your system.

jgrolik, 2023-10-06:
I first removed satdump and then started the new installation.  The installation runs successfully without any errors.  However, I did what you advised but it didn't help.  I will try to install clean OS and install NOAA-V2 again.

mihajlo2003petkovic, 2023-10-14:
> I first removed satdump and then started the new installation.  The installation runs successfully without any errors.  However, I did what you advised but it didn't help.  I will try to install clean OS and install NOAA-V2 again.

Did it work?

jgrolik, 2023-10-14:
Yes. It works. Even better. The whole problem was that RPi 2B was too slow alnd loose some data from NOAA. RPi4 works much better but I have to made better amplifier

mihajlo2003petkovic, 2023-10-14:
Yes, Pi 4 is definitely better. One suggestion is to run the Pi in headless CLI mode, so no desktop. That will definitelly sped things up and offload the CPU. I'll close the issue.

closed 2023-10-14 by mihajlo2003petkovic

jgrolik, 2023-10-14:
With RPi 4 I have NO problem with load of CPU even with VNC and desktop open. Load under 18%. With RPi 2B I had 100% with CLI. I have to move LNA with filter near to antena (today I recived Bias tee) and test gain (now it is working witch automatic gain).

mihajlo2003petkovic, 2023-10-14:
Please don't use automatic gain, try to find the best gain yourself. Automatic gain can easily create overloading and mess up your images.

jgrolik, 2023-10-14:
I know 😁 But to tests I set up auto gain. When I move LNA to antenna (as I previous indicated) I will test gain for best images. Thanks for all advices 👍