jekhokie/raspberry-noaa-v2 #806: Will not decode the Meteor M2-3

maserowik, 2023-10-09:
I am trying to decode the Meteor M2-3.  Per the log that is attached it is doing something but will not produce any images.   I jsut did a git pull yesterday and then ./install_and_upgrade.sh file still can not decode any images for Meteor M2-3  
[meteor M2-3.txt](https://github.com/jekhokie/raspberry-noaa-v2/files/12845898/meteor.M2-3.txt)

f5pql, 2023-10-09:
hi,

Works for me :

receiver_type: 'rtlsdr'
noaa_receiver: 'rtl_fm'
meteor_receiver: 'gnuradio'

maserowik, 2023-10-09:
I have both 
noaa_receiver: 'rtl_sdr'
meteor_receiver: 'rtl_sdr'

When I get back home I will change the meteor_receiver to  gnuradio and give it a try.

Mike

Francesco-Pagano, 2023-10-10:
I have the same problem, I receive the 3 noaa, but not the meteor.
I also had rtl_sdr as meteor_receiver, now I changed to gnuradio.
I'll let you know in a few hours if I received it...

Francesco-Pagano, 2023-10-12:
> i am getting a wav file but the file has not audio. Noaa 15 18 19 wav files all produce audio. Any suggestion?

Same problem

f5pql, 2023-10-13:
Normal, meteor is not analogic , it's numeric ;) no sound just a little noise with a carrier ... 

verify the log in /var/log/raspberry-noaa-v2/output.log either when passing the sat through "tail -F  /var/log/raspberry-noaa-v2/output.log" or just after passing it and processing the sound file, perhaps a library is missing, this is what happened to me and it prevents the jpg from being created correctly.

Francesco-Pagano, 2023-10-13:
This is my output of "tail -F /var/log/raspberry-noaa-v2/output.log"

Loading page (1/2)
Rendering (2/2)                                                    
Done                                                               
INFO : Extending image for annotation
13-10-2023 10:20 /home/satellite/raspberry-noaa-v2/scripts/image_processors/noaa_normalize_annotate.sh INFO : Extending image for annotation
/home/satellite/.noaa-v2.conf: riga 104: url-to-webpanel: File o directory non esistente
13-10-2023 10:20 /home/satellite/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Deleting audio files
13-10-2023 10:20 /home/satellite/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Producing polar graph of direction for pass
/home/satellite/.noaa-v2.conf: riga 104: url-to-webpanel: File o directory non esistente
13-10-2023 10:20 /home/satellite/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Total processing time: 00:23.15
13-10-2023 10:41 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : The system has enough space to store a Meteor pass on RAM
13-10-2023 10:41 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Free : 3539 ; Required : 600
13-10-2023 10:41 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Direction Southbound
13-10-2023 10:41 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Recording /home/satellite/raspberry-noaa-v2 via rtlsdr at 137.1000 MHz using GNU Radio 
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 1250000.002070 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
13-10-2023 10:57 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Removing old bmp and gcp files
13-10-2023 10:57 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Producing spectrogram
/home/satellite/.noaa-v2.conf: riga 104: url-to-webpanel: File o directory non esistente
/home/satellite/.noaa-v2.conf: riga 104: url-to-webpanel: File o directory non esistente
13-10-2023 10:58 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Running MeteorDemod to demodulate OQPSK file, rectify (spread) images, create heat map and composites and convert them to JPG
Input is a .wav file, processing it...
 Carrier: 0.00Hzz	 Lock detector: 0.00	 isLocked: 1	 OutputSize: 132.26Mb Progress: 100.00% 		

Decoded packets:0
No data received, exiting...
convert: unable to open image `spread_*.jpg': File o directory non esistente @ error/blob.c/OpenBlob/2924.
convert: no images defined `spread_*.jpg' @ error/convert.c/ConvertImageCommand/3229.
/home/satellite/.noaa-v2.conf: riga 104: url-to-webpanel: File o directory non esistente
Loading page (1/2)
Rendering (2/2)                                                    
Done                                                               
convert: unable to open image `*.jpg': File o directory non esistente @ error/blob.c/OpenBlob/2924.
convert: no images defined `/home/satellite/raspberry-noaa-v2/tmp/METEOR-M2-3-20231013-104155-*-tmp.jpg' @ error/convert.c/ConvertImageCommand/3229.
convert: unable to open image `/home/satellite/raspberry-noaa-v2/tmp/METEOR-M2-3-20231013-104155-*-tmp.jpg': File o directory non esistente @ error/blob.c/OpenBlob/2924.
convert: image sequence is required `-composite' @ error/mogrify.c/MogrifyImageList/8003.
convert: no images defined `/srv/images/METEOR-M2-3-20231013-104155-*.jpg' @ error/convert.c/ConvertImageCommand/3229.
rm: impossibile rimuovere '/home/satellite/raspberry-noaa-v2/tmp/METEOR-M2-3-20231013-104155-*-tmp.jpg': File o directory non esistente
/home/satellite/.noaa-v2.conf: riga 104: url-to-webpanel: File o directory non esistente
convert: unable to open image `/srv/images/METEOR-M2-3-20231013-104155-*.jpg': File o directory non esistente @ error/blob.c/OpenBlob/2924.
convert: no images defined `/srv/images/thumb/METEOR-M2-3-20231013-104155-*.jpg' @ error/convert.c/ConvertImageCommand/3229.
13-10-2023 11:03 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Deleting audio files
13-10-2023 11:03 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : No images found, not pushing anything
13-10-2023 11:03 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Total processing time: 00:22.11

maserowik, 2023-10-13:
What library was missing?

Sent from the all new AOL app for iOS

f5pql, 2023-10-14:
the  libopenblas was missing.

f5pql, 2023-10-14:
> This is my output of "tail -F /var/log/raspberry-noaa-v2/output.log"
>
> Loading page (1/2) Rendering (2/2) Done INFO : Extending image for annotation 13-10-2023 10:20 [...]

hi,
When reading your log, the system does not find any data to decode on your .wav file. try to copy it next time and send it to us. it should be located in /var/ramfs/meteor.*.wav.
we can confirm if it is a raw silence or the noise of a hypothetical digital sound ;)

ja-Bertolin, 2023-10-14:
Hi, in my case during the downloading it gives timeout

[10:14:46 - 14/10/2023] [33m[1m(W) Timeout is over! (943s >= 0s) Stopping.[m
[10:14:46 - 14/10/2023] [32m(I) Waiting for the thread...[m
[10:14:46 - 14/10/2023] [32m(I) Thread stopped[m
terminate called without an active exception
14-10-2023 12:14 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Removing old bmp and gcp files
14-10-2023 12:14 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Producing spectrogram
/var/ramfs/METEOR-M2-3-20231014-115947.wav: Invalid data found when processing input
14-10-2023 12:14 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Running MeteorDemod to demodulate OQPSK file, rectify (spread) images, create heat map and composites and convert them to JPG
Input is a .wav file, processing it...
Opening .wav file failed, exiting...14-10-2023 12:14 /home/pi/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Sleeping for 2 seconds to allow files to close
convert: unable to open image `spread_*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
convert: no images defined `spread_*.jpg' @ error/convert.c/ConvertImageCommand/3229.
QStandardPaths: error creating runtime directory '/run/user/1000' (Permission denied)
Loading page (1/2)
[>                                                           ] 0%
[==============================>                             ] 50%
[============================================================] 100%
Rendering (2/2)                                                    
[>                                                           ] 0%
[===============>                                            ] 25%
[============================================================] 100%
Done                                                               
convert: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
convert: no images defined `/home/pi/raspberry-noaa-v2/tmp/METEOR-M2-3-20231014-115947-*-tmp.jpg' @ error/convert.c/ConvertImageCommand/3229.
convert: unable to open image `/home/pi/raspberry-noaa-v2/tmp/METEOR-M2-3-20231014-115947-*-tmp.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
convert: image sequence is required `-composite' @ error/mogrify.c/MogrifyImageList/8003.
convert: no images defined `/srv/images/METEOR-M2-3-20231014-115947-*.jpg' @ error/convert.c/ConvertImageCommand/3229.
rm: cannot remove '/home/pi/raspberry-noaa-v2/tmp/METEOR-M2-3-20231014-115947-*-tmp.jpg': No such file or directory
convert: unable to open image `/srv/images/METEOR-M2-3-20231014-115947-*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
convert: no images defined `/srv/images/thumb/METEOR-M2-3-20231014-115947-*.jpg' @ error/convert.c/ConvertImageCommand/3229.

No audio is located in /var/ramfs/ but it is in /srv/audio/meteor
I am able to get the spectrogram only.

f5pql, 2023-10-14:
you need to copy the file right after the end of the pass while it converts it to .s

maserowik, 2023-10-14:
OK it appear a few others are having the same issue that I opened up.  I have included a file that has my METEOR-M2-3.   the files included are as follows 
1) Just the output log limited to the METEOR-M2-3.  
2) There were two files in the /var/ramfs and those files one is a *.s and the other is a *.wav. 
    METEOR-M2-3-20231014-110324.s  (130,300 KB) 
    METEOR-M2-3-20231014-110324.wav (227,303 KB) 

  These files were deleted and the only file moved to /srv/audio/meteor was METEOR-M2-3-20231014-110324.wav
     
  I have no issues with Noaa 18 or 19.  Noaa 15 I have set to false for now. 
  I did install the library as recommended above but still no METEOR-M2-3

[output.log](https://github.com/jekhokie/raspberry-noaa-v2/files/12906625/output.log)

I tried to upload the *.s and *.wav but they are too big.

Mike - N3BSQ

mihajlo2003petkovic, 2023-10-14:
> I am trying to decode the Meteor M2-3.  Per the log that is attached it is doing something but will not produce any images.   I jsut did a git pull yesterday and then ./install_and_upgrade.sh file [...]

Hello, just checked your log and you received no data from Meteor. I'd advose you to check your gain settings. Try different things. Also there is record-live-options branch which for me works even better so try switching to it.

Go to raspberry-noaa-v2 folder and run:

```bash
rm config/settings.yml
git pull
git checkout record-live-options
```
Then edit your `settings.yml` file and run the `install_and_upgrade.sh` script to update your settings.

maserowik, 2023-10-14:
> > I am trying to decode the Meteor M2-3.  Per the log that is attached it is doing something but will not produce any images.   I jsut did a git pull yesterday and then ./install_and_upgrade.sh file still can not decode any images for Meteor M2-3
> > [meteor M2-3.txt](https://github.com/jekhokie/raspberry-noaa-v2/files/12845898/meteor.M2-3.txt)
> 
> Hello, just checked your log and you received no data from Meteor. I'd advose you to check your gain settings. Try different things. Also there is record-live-options branch which for me works even better so try switching to it.
> 
> Go to raspberry-noaa-v2 folder and run:
> 
> ```shell
> rm config/settings.yml
> git pull
> git checkout record-live-options
> ```
> 
> Then edit your `settings.yml` file and run the `install_and_upgrade.sh` script to update your settings.

meteor_m2_3_schedule: true
meteor_m2_3_sdr_device_id: 0
meteor_m2_3_freq_offset: 0
meteor_m2_3_enable_bias_tee: false
meteor_m2_3_gain: 49.6
meteor_m2_3_schedule_sun_min_elevation: -90
meteor_m2_3_sun_min_elevation: 6
meteor_m2_3_sat_min_elevation: 30
meteor_m2_3_80k_interleaving: false

mihajlo2003petkovic, 2023-10-14:
Yeah, your gain seems too high. Open GQRX and wait for the next NOAA/Meteor satellite to pass and find the best gain yourself.

i6cbi, 2023-10-15:
/home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO :
 Recording /home/satellite/raspberry-noaa-v2 via rtlsdr at 137.1000 MHz
METEOR M2-3 -- 137.900   CHANGE FREQUENCY IN COMMON.SH

mihajlo2003petkovic, 2023-10-15:
> I am trying to decode the Meteor M2-3.  Per the log that is attached it is doing something but will not produce any images.   I jsut did a git pull yesterday and then ./install_and_upgrade.sh file [...]

Rum git pull again to update the frequency.

Francesco-Pagano, 2023-10-17:
> /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Recording /home/satellite/raspberry-noaa-v2 via rtlsdr at 137.1000 MHz METEOR M2-3 -- 137.900 CHANGE FREQUENCY IN COMMON.SH

I tried installing the library with "sudo apt-get install libopenblas-dev",  I changed the frequency in common.sh, and I try using the other receiver method in settings.yml
And my new output of "tail -F /var/log/raspberry-noaa-v2/output.log" is:

In the link of mega there is the .wav file ( saved in /srv/audio/meteor/ )

17-10-2023 22:10 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : The system has enough space to store a Meteor pass on RAM
17-10-2023 22:10 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Free : 3543 ; Required : 600
17-10-2023 22:10 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Direction Northbound
17-10-2023 22:10 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : I'll flip this image pass because FLIP_METEOR_IMG is set to true and PASS_DIRECTION is Northbound
17-10-2023 22:10 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Recording /home/satellite/raspberry-noaa-v2 via rtlsdr at 137.9000 MHz using GNU Radio 
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 1250000.002070 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
17-10-2023 22:25 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Removing old bmp and gcp files
17-10-2023 22:25 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Producing spectrogram
/home/satellite/.noaa-v2.conf: riga 104: url-to-webpanel: File o directory non esistente
/home/satellite/.noaa-v2.conf: riga 104: url-to-webpanel: File o directory non esistente
17-10-2023 22:26 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Running MeteorDemod to demodulate OQPSK file, rectify (spread) images, create heat map and composites and convert them to JPG
Input is a .wav file, processing it...
 Carrier: -0.00Hz	 Lock detector: 0.03	 isLocked: 1	 OutputSize: 126.36Mb Progress: 100.00% 

Decoded packets:0
No data received, exiting...
convert: unable to open image `spread_*.jpg': File o directory non esistente @ error/blob.c/OpenBlob/2924.
convert: no images defined `spread_*.jpg' @ error/convert.c/ConvertImageCommand/3229.
/home/satellite/.noaa-v2.conf: riga 104: url-to-webpanel: File o directory non esistente
Loading page (1/2)
Rendering (2/2)                                                    
Done                                                               
convert: unable to open image `*.jpg': File o directory non esistente @ error/blob.c/OpenBlob/2924.
convert: no images defined `/home/satellite/raspberry-noaa-v2/tmp/METEOR-M2-3-20231017-221040-*-tmp.jpg' @ error/convert.c/ConvertImageCommand/3229.
convert: unable to open image `/home/satellite/raspberry-noaa-v2/tmp/METEOR-M2-3-20231017-221040-*-tmp.jpg': File o directory non esistente @ error/blob.c/OpenBlob/2924.
convert: image sequence is required `-composite' @ error/mogrify.c/MogrifyImageList/8003.
convert: no images defined `/srv/images/METEOR-M2-3-20231017-221040-*.jpg' @ error/convert.c/ConvertImageCommand/3229.
rm: impossibile rimuovere '/home/satellite/raspberry-noaa-v2/tmp/METEOR-M2-3-20231017-221040-*-tmp.jpg': File o directory non esistente
/home/satellite/.noaa-v2.conf: riga 104: url-to-webpanel: File o directory non esistente
convert: unable to open image `/srv/images/METEOR-M2-3-20231017-221040-*.jpg': File o directory non esistente @ error/blob.c/OpenBlob/2924.
convert: no images defined `/srv/images/thumb/METEOR-M2-3-20231017-221040-*.jpg' @ error/convert.c/ConvertImageCommand/3229.
17-10-2023 22:31 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Moving audio files out to the SD card
17-10-2023 22:31 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Deleting Meteor audio files older than 30 days
17-10-2023 22:31 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : No images found, not pushing anything
17-10-2023 22:31 /home/satellite/raspberry-noaa-v2/scripts/receive_meteor.sh INFO : Total processing time: 00:21.22

[METEOR-M2-3-20231017-221040.wav]
https://mega.nz/file/bGICWQKD#Jnn87jTL9MFLmG8o70Gp-zgW9Lnnf2t78VnTPu1AdrY

Francesco-Pagano, 2023-11-11:
I downloaded the latest version from the site https://drive.google.com/drive/folders/1nOycJlummsoaRO2gIdtVjLGzVjke_KTs (Build 10) and I was finally able to capture the Meteor images!!!

mbruski, 2023-11-16:
@Francesco-Pagano How did you unzip this file?  I get an error in balena-etcher saying the compression format is unsupported.  I also receive an error using unzip and xarchiver.

mbruski, 2023-11-16:
Sorry, never-mind.   I found the 7zip will unravel it.

maserowik, 2023-11-20:
I had to set my gain to 40.2 and bingo i am now able to get Meteor M2-3
Closing

closed 2023-11-20 by maserowik