jekhokie/raspberry-noaa-v2 #842: install_and_upgrade.sh run into error and does not complete

dc5dm, 2023-10-15:
Hi, I am stuck with the installation. 

After installing all dependices, the "script install_and_upgrade.sh" run into an error.
Console output of the error:
```
[...]
TASK [common : Installing AnyDesk remote desktop] ******************************************************************
ok: [localhost]

TASK [common : noaa environment variables] *************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "checksum": "aa5f563f9a537cffb7ee5a9fd0966f079ef53a64", "gid": 100, "group": "users", "mode": "0644", "msg": "chgrp failed", "owner": "marcus", "path": "/home/marcus/.noaa-v2.conf", "size": 3581, "state": "file", "uid": 1000}

PLAY RECAP *********************************************************************************************************
localhost                  : ok=10   changed=0    unreachable=0    failed=1    skipped=11   rescued=0    ignored=0   

Error:   Something failed with the install - please inspect the logs above
```

**Support Output**
```
=============================================
Details about environment
=============================================
Current local date/time: So 15. Okt 14:52:01 CEST 2023
Current date/time (ms):  1697374321
Repo git hash:           5624adb796dcb47ff9bbbedfcbb7cb15c97f3b79
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.5
Architecture:            Linux rpinoaa 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  (None)
---------------------------------------------
Satellite SDR Settings:
  meteor_m2_memory_threshold: 600
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: 0
  noaa_15_enable_bias_tee: false
  noaa_15_gain: 29.7
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 30
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: 0
  noaa_18_enable_bias_tee: false
  noaa_18_gain: 29.7
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 30
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: 0
  noaa_19_enable_bias_tee: false
  noaa_19_gain: 29.7
  noaa_19_sun_min_elevation: 6
  noaa_19_sat_min_elevation: 30
  meteor_m2_3_schedule: true
  meteor_m2_3_sdr_device_id: 0
  meteor_m2_3_freq_offset: 0
  meteor_m2_3_enable_bias_tee: false
  meteor_m2_3_gain: 29.7
  meteor_m2_3_schedule_sun_min_elevation: -90
  meteor_m2_3_sun_min_elevation: 6
  meteor_m2_3_sat_min_elevation: 30
  meteor_m2_3_80k_interleaving: false
---------------------------------------------
USB Device Map:
  Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
  Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
  Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
  mmcblk0     179:0    0 238,3G  0 disk 
  ├─mmcblk0p1 179:1    0   256M  0 part /boot
  └─mmcblk0p2 179:2    0   238G  0 part /
---------------------------------------------
Disk Usage Info:
  Dateisystem    1K-Blöcke Benutzt Verfügbar Verw% Eingehängt auf
  /dev/root      245569988 8014504 227476740    4% /
  devtmpfs          682576       0    682576    0% /dev
  tmpfs             946800       0    946800    0% /dev/shm
  tmpfs             378720     964    377756    1% /run
  tmpfs               5120       4      5116    1% /run/lock
  tmpfs            1024000       0   1024000    0% /var/ramfs
  /dev/mmcblk0p1    258095   51409    206687   20% /boot
  tmpfs             189360      20    189340    1% /run/user/109
  tmpfs             189360      20    189340    1% /run/user/1000
---------------------------------------------
Memory Info:
  MemTotal:        1893600 kB
  MemFree:          237468 kB
  MemAvailable:    1644348 kB
---------------------------------------------
Git source files changed:
  config/settings.yml
---------------------------------------------
SDR Information:
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
---------------------------------------------
Database tables:
  (None)
```

**Server information (please complete the following information):**
 - OS: debian 11.7
 - Python version 3.9.2
 - SDR device Nooelec rtlsdr

I am stuck installing and running the system and have no idea how to proceed and fix from here.
Help/hints how to solve appreciated.

mni tnx
Marcus

smarti2019, 2023-10-15:
easy, install anydesk original Package from:
https://anydesk.com/de/downloads/raspberry-pi?dv=raspberrypi
download the deb package and then install with
sudo apt-get install  xxxx.deb

selfroasted, 2023-10-16:
Just out of curiosity, why is Anydesk part of the mandatory installation for something many are running headless?

dc5dm, 2023-10-16:
Hi, thanks for reply. However, I think it isn't as easy as thought...
```marcus@rpinoaa:~ $ sudo apt-get install ./anydesk_6.3.0-1_armhf.deb 
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Hinweis: »anydesk« wird an Stelle von »./anydesk_6.3.0-1_armhf.deb« gewählt.
anydesk ist schon die neueste Version (6.3.0).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 89 nicht aktualisiert.
```
Semms the most recent anydesk was already installed.
The script "install_and_upgrade.sh" seems to run some steps further
```
TASK [common : wxtoimg config file] ************************************************************************************************
changed: [localhost]

TASK [common : MeteorDemod config settings.ini file] *******************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "checksum": "455c6cca060a5e3c14a03824191a7b5a63028812", "msg": "Destination directory /home/marcus/.config/meteordemod does not exist"}

PLAY RECAP *************************************************************************************************************************
localhost                  : ok=14   changed=4    unreachable=0    failed=1    skipped=11   rescued=0    ignored=0   

Error:   Something failed with the install - please inspect the logs above
```
I created the missing directory by hand and the script ran until here
``` * Updating database schema with any changes...
/home/marcus/.noaa-v2.conf: Zeile 104: url-to-webpanel: Datei oder Verzeichnis nicht gefunden
INFO : Applying any schema updates...
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
INFO :   - applying 00_seed_schema.sql
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
INFO :   - applying 01_add_spectrogram_bool.sql
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
INFO :   - applying 02_add_noaa_pristine_bool.sql
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
INFO :   - applying 03_add_pass_azimuth_direction.sql
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
INFO :   - applying 04_add_capture_gain.sql
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
INFO :   - applying 05_add_pass_azimuth_at_max.sql
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
INFO :   - applying 06_add_polar_az_el_bool.sql
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
INFO :   - applying 07_add_polar_direction_bool.sql
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
INFO :   - applying 08_add_histogram_bool.sql
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
INFO :   - applying 09_add_at_job_id.sql
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
INFO : Schema updates complete!
/home/marcus/raspberry-noaa-v2/scripts/common.sh: Zeile 23: /var/log/raspberry-noaa-v2/output.log: Keine Berechtigung
Error:   Something failed with database update - please inspect the logs above
```
For this to fix I set 777 as access right for the directory "/var/log/raspberry-noaa-v2 and to delete old output.log files from previous tries.

After this the script ran through. Thanks so far. I am courious to see first rx results.

73 de dc5dm, Marcus

MihajloPi, 2024-12-02:
Anydesk was removed and this issue is no longer relevant.

closed 2024-12-02 by MihajloPi