jekhokie/raspberry-noaa-v2 #843: [Error] libopenblas-base wont install

bitliandenis, 2023-10-18:
**Describe the bug**
im trying to install raspberry noaa v2 on my raspberry pi 400 (running 32 bit version for compability) and when i run the install script, i just get this error:

**Screenshots and/or logs**
If applicable, add screenshots to help explain your problem. Logs are always welcome as well.

```
PLAY [install raspberry-noaa-v2 core] **********************************************************************************
TASK [get os release] **************************************************************************************************ok: [localhost]

TASK [common : install dependencies] ***********************************************************************************fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'libopenblas-base' is available"}

PLAY RECAP *************************************************************************************************************localhost                  : ok=1    changed=0    unreachable=0    failed=1    skipped=0    rescued=0    ignored=0

Error:   Something failed with the install - please inspect the logs above
```

**Support Output**
```
=============================================
Details about environment
=============================================
Current local date/time: Wed 18 Oct 21:07:54 EEST 2023
Current date/time (ms):  1697652474
Repo git hash:           5624adb796dcb47ff9bbbedfcbb7cb15c97f3b79
Raspberry Pi Model:      Raspberry Pi 400 Rev 1.1
Architecture:            Linux raspberrypi 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
./support.sh: line 36: atq: command not found
  (None)
---------------------------------------------
Satellite SDR Settings:
  meteor_m2_memory_threshold: 600
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: 0
  noaa_15_enable_bias_tee: false
  noaa_15_gain: 29.7
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 30
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: 0
  noaa_18_enable_bias_tee: false
  noaa_18_gain: 29.7
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 30
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: 0
  noaa_19_enable_bias_tee: false
  noaa_19_gain: 29.7
  noaa_19_sun_min_elevation: 6
  noaa_19_sat_min_elevation: 30
  meteor_m2_3_schedule: true
  meteor_m2_3_sdr_device_id: 0
  meteor_m2_3_freq_offset: 0
  meteor_m2_3_enable_bias_tee: false
  meteor_m2_3_gain: 29.7
  meteor_m2_3_schedule_sun_min_elevation: -90
  meteor_m2_3_sun_min_elevation: 6
  meteor_m2_3_sat_min_elevation: 30
  meteor_m2_3_80k_interleaving: false
---------------------------------------------
USB Device Map:
  Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
  Bus 001 Device 004: ID 04d9:0007 Holtek Semiconductor, Inc. Raspberry Pi Internal Keyboard
  Bus 001 Device 003: ID 145f:021f Trust 2.4G Mouse
  Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
  mmcblk0     179:0    0 58.2G  0 disk
  ├─mmcblk0p1 179:1    0  512M  0 part /boot/firmware
  └─mmcblk0p2 179:2    0 57.7G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks    Used Available Use% Mounted on
  udev             1670004       0   1670004   0% /dev
  tmpfs             388424    1020    387404   1% /run
  /dev/mmcblk0p2  59523636 4941656  51538552   9% /
  tmpfs            1942116     268   1941848   1% /dev/shm
  tmpfs               5120      12      5108   1% /run/lock
  /dev/mmcblk0p1    522232  115288    406944  23% /boot/firmware
  tmpfs             388420      36    388384   1% /run/user/1000
---------------------------------------------
Memory Info:
  MemTotal:        3884236 kB
  MemFree:         2957572 kB
  MemAvailable:    3517656 kB
---------------------------------------------
Git source files changed:
  (None)
---------------------------------------------
SDR Information:
./support.sh: line 14: rtl_eeprom: command not found
---------------------------------------------
Database tables:
  (None).

**Server information (please complete the following information):**
 - OS: [e.g. Raspbian]
 - Python version [e.g. 3.7.3]
 - SDR device [e.g. RTL-SDR v3]
```

**Additional context**
clean raspberry pi installation, had locale settings before but fixed

bitliandenis, 2023-10-19:
in additional context, i meant i had an error with locale settings

kiwikierab, 2023-12-06:
What version of Raspbian are you running? The latest once has this issue among many others. Change it to Bullseye Lite 32Bit and all the issues should away.

mattfox27, 2024-01-29:
I did bullseye light 32 and same error

MihajloPi, 2024-12-02:
Latest releases do not feature this error.

closed 2024-12-02 by MihajloPi