jekhokie/raspberry-noaa-v2 #854: ./install_and_upgrade.sh fails on Raspberry Pi 4 Model B Rev 1.2

gjmkoper, 2023-10-22:
**Describe the bug**
The ./install_and_upgrade.sh fails on Raspberry Pi 4 Model B Rev 1.2

**Screenshots and/or logs**
TASK [common : Installing WXtoImg] *********************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Unable to install package: E:Could not open file /home/pi/raspberry-noaa-v2/software/wxtoimg-arm64-2.11.2-beta.deb - open (2: No such file or directory)"}

**Support Output**
=============================================
Details about environment
=============================================
Current local date/time: Sun 22 Oct 2023 07:09:32 PM CEST
Current date/time (ms):  1697994572
Repo git hash:           5624adb796dcb47ff9bbbedfcbb7cb15c97f3b79
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.2
Architecture:            Linux SatPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  (None)
---------------------------------------------
Satellite SDR Settings:
  meteor_m2_memory_threshold: 600
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: 0
  noaa_15_enable_bias_tee: false
  noaa_15_gain: 29.7
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 30
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: 0
  noaa_18_enable_bias_tee: false
  noaa_18_gain: 29.7
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 30
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: 0
  noaa_19_enable_bias_tee: false
  noaa_19_gain: 29.7
  noaa_19_sun_min_elevation: 6
  noaa_19_sat_min_elevation: 30
  meteor_m2_3_schedule: true
  meteor_m2_3_sdr_device_id: 0
  meteor_m2_3_freq_offset: 0
  meteor_m2_3_enable_bias_tee: false
  meteor_m2_3_gain: 29.7
  meteor_m2_3_schedule_sun_min_elevation: -90
  meteor_m2_3_sun_min_elevation: 6
  meteor_m2_3_sat_min_elevation: 30
  meteor_m2_3_80k_interleaving: false
---------------------------------------------
USB Device Map:
  Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
  Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
  Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
  mmcblk0     179:0    0 59.5G  0 disk
  ├─mmcblk0p1 179:1    0  256M  0 part /boot
  └─mmcblk0p2 179:2    0 59.2G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks    Used Available Use% Mounted on
  /dev/root       61052320 6504256  52008076  12% /
  devtmpfs         1678472       0   1678472   0% /dev
  tmpfs            1942696       0   1942696   0% /dev/shm
  tmpfs             777080    1280    775800   1% /run
  tmpfs               5120       4      5116   1% /run/lock
  /dev/mmcblk0p1    261108   31374    229734  13% /boot
  tmpfs             388536      24    388512   1% /run/user/1000
---------------------------------------------
Memory Info:
  MemTotal:        3885396 kB
  MemFree:         2879312 kB
  MemAvailable:    3472476 kB
---------------------------------------------
Git source files changed:
  config/settings.yml
---------------------------------------------
SDR Information:
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:              0x0bda
Product ID:             0x2838
Manufacturer:           Realtek
Product:                RTL2838UHIDIR
Serial number:          00000001
Serial number enabled:  yes
IR endpoint enabled:    yes
Remote wakeup enabled:  no
__________________________________________
Reattached kernel driver
---------------------------------------------
Database tables:
  (None)

**Server information (please complete the following information):**
 See support info above.

**Additional context**
Not available.

mbruski, 2023-10-29:
I experienced the same problem this morning.  Got around it by creating a soft link in the software directory.

cd ~/raspberry-noaa-v2/software/
ln -s wxtoimg-armhf-2.11.2-beta.deb wxtoimg-arm64-2.11.2-beta.deb

This fixed that issue but then I encountered a similar issue with satdump.   So I created another soft link...

ln -s satdump_1.1.1-d649918b_armhf.deb satdump_1.1.1-d649918b_arm64.deb

Unfortunately that didn't work as I reaped a number of dependency issues as a result.   Don't know what to do now.

I'm installing on Raspberry Pi 4 4GB, OS bullseye as none 'pi' user.

TASK [common : Installing SatDump] ******************************************************************************************************************************************************************************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "dpkg --force-confdef --force-confold -i /home/aj9x/raspberry-noaa-v2/software/satdump_1.1.1-d649918b_arm64.deb failed", "stderr": "dpkg: dependency problems prevent configuration of satdump:armhf:\n satdump:armhf depends on ocl-icd-libopencl1 | libopencl-1.1-1 | libopencl1.\n satdump:armhf depends on ocl-icd-libopencl1 | libopencl1.\n\ndpkg: error processing package satdump:armhf (--install):\n dependency problems - leaving unconfigured\nErrors were encountered while processing:\n satdump:armhf\n", "stderr_lines": ["dpkg: dependency problems prevent configuration of satdump:armhf:", " satdump:armhf depends on ocl-icd-libopencl1 | libopencl-1.1-1 | libopencl1.", " satdump:armhf depends on ocl-icd-libopencl1 | libopencl1.", "", "dpkg: error processing package satdump:armhf (--install):", " dependency problems - leaving unconfigured", "Errors were encountered while processing:", " satdump:armhf"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following additional packages will be installed:\n  gcc-10-base:armhf krb5-locales libatomic1:armhf libbsd0:armhf\n  libcom-err2:armhf libcrypt1:armhf libdrm-amdgpu1:armhf libdrm-nouveau2:armhf\n  libdrm-radeon1:armhf libdrm2:armhf libedit2:armhf libelf1:armhf\n  libexpat1:armhf libffi7:armhf libgl1-mesa-dri:armhf libglapi-mesa:armhf\n  libglvnd0:armhf libglx-mesa0:armhf libglx0:armhf libgssapi-krb5-2:armhf\n  libidn2-0:armhf libk5crypto3:armhf libkeyutils1:armhf libkrb5-3:armhf\n  libkrb5support0:armhf libllvm11:armhf libmd0:armhf libnsl2:armhf\n  libnss-nis:armhf libnss-nisplus:armhf liborc-0.4-0:armhf libsensors5:armhf\n  libssl1.1:armhf libtinfo6:armhf libtirpc3:armhf libudev1:armhf\n  libunistring2:armhf libvulkan1:armhf libwayland-client0:armhf libx11-6:armhf\n  libx11-xcb1:armhf libxau6:armhf libxcb-dri2-0:armhf libxcb-dri3-0:armhf\n  libxcb-glx0:armhf libxcb-present0:armhf libxcb-randr0:armhf\n  libxcb-shm0:armhf libxcb-sync1:armhf libxcb-xfixes0:armhf libxcb1:armhf\n  libxdamage1:armhf libxdmcp6:armhf libxext6:armhf libxfixes3:armhf\n  libxshmfence1:armhf libxxf86vm1:armhf libz3-4:armhf\n  mesa-vulkan-drivers:armhf\nSuggested packages:\n  glibc-doc:armhf locales:armhf libfftw3-dev:armhf glew-utils:armhf\n  libegl1:armhf libgles1:armhf libgles2:armhf libosmesa6:armhf krb5-doc:armhf\n  krb5-user:armhf lm-sensors:armhf libvolk2-dev:armhf\nRecommended packages:\n  libvolk2-bin:armhf\nThe following NEW packages will be installed:\n  gcc-10-base:armhf krb5-locales libairspy0:armhf libairspyhf1:armhf\n  libatomic1:armhf libbsd0:armhf libc6:armhf libcom-err2:armhf libcrypt1:armhf\n  libdrm-amdgpu1:armhf libdrm-nouveau2:armhf libdrm-radeon1:armhf\n  libdrm2:armhf libedit2:armhf libelf1:armhf libexpat1:armhf libffi7:armhf\n  libfftw3-single3:armhf libgcc-s1:armhf libgl1:armhf libgl1-mesa-dri:armhf\n  libglapi-mesa:armhf libglew2.1:armhf libglfw3:armhf libglvnd0:armhf\n  libglx-mesa0:armhf libglx0:armhf libgomp1:armhf libgssapi-krb5-2:armhf\n  libhackrf0:armhf libidn2-0:armhf libk5crypto3:armhf libkeyutils1:armhf\n  libkrb5-3:armhf libkrb5support0:armhf libllvm11:armhf libluajit-5.1-2:armhf\n  libmd0:armhf libnng1:armhf libnsl2:armhf libnss-nis:armhf\n  libnss-nisplus:armhf liborc-0.4-0:armhf libpng16-16:armhf librtlsdr0:armhf\n  libsensors5:armhf libssl1.1:armhf libstdc++6:armhf libtinfo6:armhf\n  libtirpc3:armhf libudev1:armhf libunistring2:armhf libusb-1.0-0:armhf\n  libvolk2.4:armhf libvulkan1:armhf libwayland-client0:armhf libx11-6:armhf\n  libx11-xcb1:armhf libxau6:armhf libxcb-dri2-0:armhf libxcb-dri3-0:armhf\n  libxcb-glx0:armhf libxcb-present0:armhf libxcb-randr0:armhf\n  libxcb-shm0:armhf libxcb-sync1:armhf libxcb-xfixes0:armhf libxcb1:armhf\n  libxdamage1:armhf libxdmcp6:armhf libxext6:armhf libxfixes3:armhf\n  libxshmfence1:armhf libxxf86vm1:armhf libz3-4:armhf libzstd1:armhf\n  mesa-vulkan-drivers:armhf zlib1g:armhf\n0 upgraded, 78 newly installed, 0 to remove and 0 not upgraded.\nNeed to get 41.5 MB of archives.\nAfter this operation, 155 MB of additional disk space will be used.\nGet:1 http://deb.debian.org/debian bullseye/main armhf gcc-10-base armhf 10.2.1-6 [201 kB]\nGet:2 http://security.debian.org/debian-security bullseye-security/main armhf libx11-6 armhf 2:1.7.2-1+deb11u2 [713 kB]\nGet:3 http://deb.debian.org/debian bullseye/main arm64 krb5-locales all 1.18.3-6+deb11u4 [95.7 kB]\nGet:4 http://deb.debian.org/debian bullseye/main armhf libgcc-s1 armhf 10.2.1-6 [36.2 kB]\nGet:5 http://deb.debian.org/debian bullseye/main armhf libcrypt1 armhf 1:4.4.18-4 [... cut ...]
PLAY RECAP ******************************************************************************************************************************************************************************************************************************************************************************************************************
localhost                  : ok=5    changed=0    unreachable=0    failed=1    skipped=10   rescued=0    ignored=0   

Error:   Something failed with the install - please inspect the logs above
user@host:~/raspberry-noaa-v2 $

mbruski, 2023-10-29:
The satdump error I encountered is easily fixed running the following command:

sudo apt --fix-broken

That said, anydesk has the same problem.   Create the following soft link:

ln -s anydesk_6.3.0-1_armhf.deb  anydesk_6.3.0-1_arm64.deb

and rerun ./install_and_upgrade.sh but yet more errors crop up and --fix-broken  doesn't resolve the issue.  Stuck again.

mbruski, 2023-10-29:
Is anydesk really neaded to run RN2?  If not, can it be deleted as a dependency?  I can not even install it manually with dpkg.  It keeps throwing post-processing errors.

mihajlo2003petkovic, 2023-10-30:
What OS version are you running? 64 bit or 32 bit? What Raspbian version? Bullseye or Bookworm? RN2 currently only works with 32 bit Bullseye. It doesn't work on Bookworm or on 64 bit OS.

gjmkoper, 2023-10-30:
I gave all information you asked for. I got the impression that this version would run with the 64 bit OS. Apparently not (yet).

closed 2023-10-30 by gjmkoper

mattfox27, 2024-02-11:
im trying to install on Bullseye 32 bit and keep getting an error about anydesk as well, how can I fix

jrjstar, 2024-02-26:
Also try the app image