jekhokie/raspberry-noaa-v2 #866: public web site

maserowik, 2023-11-14:
Is there a way to upload the raspberry-noaa local website to a public hosting site?

Mike

maserowik, 2024-02-15:
@Lopastudio 

> There is no automated way, as from what I know. But you can copy the images from `/srv/images` and manually copy them to your ftp website. If you want, I can make you a shell script.

That would be fantastic.  Thanks for the offer.

Lopastudio, 2024-02-23:
# 1. Installing the script

First things first, change the directory to your home (or where you want the script to exist):
```sh
cd /home/patrik
```
You should, obviously, replace the username (mine is Patrik so...)

Next, paste the script into a file, using NANO:
```sh
nano uploader.sh
```
it will open up a text editor, where you need to paste the following script:
```sh
#!/bin/bash

#FTP credentials/details
FTP_SERVER="ftp_address_here (ftp.example.org)"
FTP_USERNAME="user_here"
FTP_PASSWORD="password_here"

#Local directory to upload
IMAGES_DIR="/srv/images"
REMOTE_DIR="/REPLACE_ME_WITH_YOUR_PATH_PLZ"

#Connect to FTP server
ftp -n $FTP_SERVER <<END_SCRIPT
quote USER $FTP_USERNAME
quote PASS $FTP_PASSWORD
cd $REMOTE_DIR
binary
lcd $IMAGES_DIR
mput *

quit
END_SCRIPT
```
next, to exit and save nano, just press CTRL + X, then Y and finally enter to save (thanks, networkchuck)

You also need to add a permission to this script, to be executable. That can be done with this:
```sh
chmod +x uploader.sh
```
And we are finished. Just run this script and you are good to go. But it is not automatic, to make it automatic, there is the next chapter.

# 2. Making it automatic (optional)

If you want to make this script run automatically, we´ll use crontab. 
To open the crontab config, you need to run this command:
```sh
crontab -e
```
choose your editor of choice (I recommend nano, because it is the easiest) and add the following text to the bottom of the file:

```
# Automatic FTP uploader for images
0 0 * * * /home/patrik/uploader.sh 
```
Of course, you will need to change the user directory name (mine is Patrik) to yours. Or change the path to any path, where you installed the script in chapter 1.

And you are good to go. To finalize, just restart your RPi with a simple command and the script should run every 24 hours, and automatically upload.
```sh
sudo reboot
```

## I recommend running the script manually, to see if it works, to save you time and headaces.