jekhokie/raspberry-noaa-v2 #887: [Error] Raspberry Pi 5 - installation error

lgmendes, 2023-12-16:
when I run ./install_and_upgrade.sh I get a error related with "No package matching 'libopenblas-base' 
on a Raspberry Pi 5 (Debian GNU/Linux 12 (bookworm))

'''
lgmendes@raspberrypi5:~ $ cd raspberry-noaa-v2/
lgmendes@raspberrypi5:~/raspberry-noaa-v2 $ ./install_and_upgrade.sh
 * Installing yaml and jsonschema Python modules...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-yaml is already the newest version (6.0-3+b2).
python3-jsonschema is already the newest version (4.10.3-1).
The following package was automatically installed and is no longer required:
  rtimucli
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
 * Checking configuration files...
SUCCESS: The configuration config/settings.yml conforms to the expected schema!
 ✓   Config check complete!
 * Checking for configuration settings...
 ✓   Settings file ready!
 * Running Ansible to install and/or update your raspberry-noaa-v2...
[DEPRECATION WARNING]: "include" is deprecated, use include_tasks/import_tasks 
instead. See https://docs.ansible.com/ansible-
core/2.14/user_guide/playbooks_reuse_includes.html for details. This feature 
will be removed in version 2.16. Deprecation warnings can be disabled by 
setting deprecation_warnings=False in ansible.cfg.

PLAY [install raspberry-noaa-v2 core] ******************************************

TASK [get os release] **********************************************************
ok: [localhost]

TASK [common : install dependencies] *******************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'libopenblas-base' is available"}

PLAY RECAP *********************************************************************
localhost                  : ok=1    changed=0    unreachable=0    failed=1    skipped=0    rescued=0    ignored=0   

Error:   Something failed with the install - please inspect the logs above
'''

Tomrm96, 2023-12-24:
Same here, dissapointing doesn't look like it's going to get fixed, there has been no activity in a while :(

jp112sdl, 2023-12-27:
Try it with `- libblas3` instead of `- libopenblas-base`
https://github.com/jekhokie/raspberry-noaa-v2/blob/6312b18ffccb6bc8e6d4c8d048cc8d658de9355d/ansible/roles/common/tasks/dependencies.yml#L24
with

lgmendes, 2023-12-27:
Thank! It solves the problem. 
However I found other problem related with Python and Virtual environments: 

TASK [common : Install required Python packages using pip3] **********************************************************************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "cmd": ["/usr/bin/pip3", "install", "envbash==1.2.0", "facebook-sdk==3.1.0", "pysqlite3==0.5.1"], "msg": "\n:stderr: error: externally-managed-environment\n\n× This environment is externally managed\n╰─> To install Python packages system-wide, try apt install\n    python3-xyz, where xyz is the package you are trying to\n    install.\n    \n    If you wish to install a non-Debian-packaged Python package,\n    create a virtual environment using python3 -m venv path/to/venv.\n    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make\n    sure you have python3-full installed.\n    \n    For more information visit http://rptl.io/venv\n\nnote: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.\nhint: See PEP 668 for the detailed specification.\n"}

-----------------------------------
The full output:

"""lgmendes@raspberrypi5:~/raspberry-noaa-v2 $ ./install_and_upgrade.sh
 * Installing yaml and jsonschema Python modules...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-yaml is already the newest version (6.0-3+b2).
python3-jsonschema is already the newest version (4.10.3-1).
The following package was automatically installed and is no longer required:
  rtimucli
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
 * Checking configuration files...
SUCCESS: The configuration config/settings.yml conforms to the expected schema!
 ✓   Config check complete!
 * Checking for configuration settings...
 ✓   Settings file ready!
 * Running Ansible to install and/or update your raspberry-noaa-v2...
[DEPRECATION WARNING]: "include" is deprecated, use include_tasks/import_tasks instead. See https://docs.ansible.com/ansible-core/2.14/user_guide/playbooks_reuse_includes.html for details. This feature will be removed in version 2.16. 
Deprecation warnings can be disabled by setting deprecation_warnings=False in ansible.cfg.

PLAY [install raspberry-noaa-v2 core] ********************************************************************************************************************************************************************************************************

TASK [get os release] ************************************************************************************************************************************************************************************************************************
ok: [localhost]

TASK [common : install dependencies] *********************************************************************************************************************************************************************************************************
ok: [localhost]

TASK [common : Install required Python packages using pip3] **********************************************************************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "cmd": ["/usr/bin/pip3", "install", "envbash==1.2.0", "facebook-sdk==3.1.0", "pysqlite3==0.5.1"], "msg": "\n:stderr: error: externally-managed-environment\n\n× This environment is externally managed\n╰─> To install Python packages system-wide, try apt install\n    python3-xyz, where xyz is the package you are trying to\n    install.\n    \n    If you wish to install a non-Debian-packaged Python package,\n    create a virtual environment using python3 -m venv path/to/venv.\n    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make\n    sure you have python3-full installed.\n    \n    For more information visit http://rptl.io/venv\n\nnote: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.\nhint: See PEP 668 for the detailed specification.\n"}

PLAY RECAP ***********************************************************************************************************************************************************************************************************************************
localhost                  : ok=2    changed=0    unreachable=0    failed=1    skipped=0    rescued=0    ignored=0   

Error:   Something failed with the install - please inspect the logs above

"""

Tomrm96, 2023-12-27:
Looks to be the same with me:

[image]

Tomrm96, 2023-12-27:
Seems to be a change in the way you install things, I've installed all 3 manually in a vertical env as suggested here https://stackoverflow.com/questions/75602063/pip-install-r-requirements-txt-is-failing-this-environment-is-externally-mana

But I still get the error 

[image]
****

Tomrm96, 2023-12-27:
I think some tweeking of this is involved 

- name: Install required Python packages using pip3
  pip:
    name:
      - envbash==1.2.0
      - facebook-sdk==3.1.0
      - pysqlite3==0.5.1
https://github.com/jekhokie/raspberry-noaa-v2/blob/6312b18ffccb6bc8e6d4c8d048cc8d658de9355d/ansible/roles/common/tasks/dependencies.yml#L63

but I'm not sure how

Tomrm96, 2023-12-27:
use:

pip3 install --break-system-packages pysqlite3 

replace pysql with the other packages. 

I managed to install them 
[image]

but when re-running it still tries to install them and throws the same error. Lets wait for Jerome. 
[image]

jp112sdl, 2023-12-27:
The only package I installed with `--break-system-packages` was `envbash`.

But I came from a Raspberry Pi 4 with Bullseye and RN2 installed (long time ago). 
Then I did an dist-upgrade to Bookworm and moved the sd card to the Raspberry Pi 5.

Tomrm96, 2023-12-27:
Yeah I think there has been changes made, I think the script might need changing, It possibly doesn't look to see if they are already installs and then trys to install them anyway which doesn't work due to the recent change

ecriscente, 2024-01-26:
`sudo rm /usr/lib/python3.11/EXTERNALLY-MANAGED`

yellowcooln, 2024-02-28:
I am doing this on the following version on a Raspberry Pi 4 2GB - Pi OS Lite (64-bit). _I know this thread is about rpi5, but I wanted to try and help with my testing._ 
```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
```
Doing everything above. 

> Try it with `- libblas3` instead of `- libopenblas-base`
> 
> https://github.com/jekhokie/raspberry-noaa-v2/blob/6312b18ffccb6bc8e6d4c8d048cc8d658de9355d/ansible/roles/common/tasks/dependencies.yml#L24
> 
> with

```
pip3 install --break-system-packages pysqlite3
pip3 install --break-system-packages envbash
pip3 install --break-system-packages facebook-sdk
```

> `sudo rm /usr/lib/python3.11/EXTERNALLY-MANAGED`

This did get us further. 

I am going to log everything I've done to get to the next road block. 

[image]

```
cd raspberry-noaa-v2/software/
mv wxtoimg-armhf-2.11.2-beta.deb wxtoimg-arm64-2.11.2-beta.deb
cd ..
./install_and_upgrade.sh
```
[image]
```
cd raspberry-noaa-v2/software/
mv satdump_1.1.4-1ccc628f_armhf.deb satdump_1.1.4-1ccc628f_arm64.deb
cd ..
./install_and_upgrade.sh
```
[image]

```
cd raspberry-noaa-v2/software/
mv anydesk_6.3.0-1_armhf.deb anydesk_6.3.0-1_arm64.deb
cd ..
./install_and_upgrade.sh
```
This next crash is so bad that I'm not posting it on pastebin [https://pastebin.com/LhmejmNR](https://pastebin.com/LhmejmNR)

hamdyaea, 2024-03-05:
I just had the same kinds of problems on a 32 bit version of Raspbian Bookworm on a Raspberry pi 4. I think that this scripts have been made for Raspbian bullseye but produce errors on Bookworms. Finally I have installed a prebuild image on my micro sd card..

hahnx117, 2024-06-07:
The problem here is the architecture. There is no arm64 version of the packages it's trying to install in the software directory, so it's bombing.

[Issue 463 ](https://github.com/jekhokie/raspberry-noaa-v2/issues/463) references installing packages for armhf to arm64, however since you have to use `dpkg` it can't install dependencies, and you have to do that by hand. If you try to install them with `dpkg -i` you'll get 

```
 satdump:armhf depends on libvolk-bin | libvolk2-bin.
 satdump:armhf depends on libnng1.
 satdump:armhf depends on librtlsdr0 | librtlsdr2.
 satdump:armhf depends on libhackrf0.
 satdump:armhf depends on libairspy0.
 satdump:armhf depends on libairspyhf1.
 satdump:armhf depends on libglfw3 | libglfw3-wayland.
 satdump:armhf depends on libzstd1.
 satdump:armhf depends on libjemalloc2 | libjemalloc1.
```
You can install those, but you need to add `:armhf` after. I did this after going through the steps for Issue 463, so I don't know if it depends on that.

I've forked this repo to see if I can build a switch into the Ansible roles to use this workaround. I was able to install `wxtoimg` as well as `satdump` by manually installing the dependencies, but had trouble with the `anydesk` service The dependencies install, but I get the following error:

```
Jun 07 10:36:54 pi5-noaa anydesk[829]: /usr/bin/anydesk: error while loading shared libraries: libatk-1.0.so.0: ELF load command address/offset not page-aligned
```

MihajloPi, 2024-12-02:
Raspberry Pi 5 is fully supported now.

closed 2024-12-02 by MihajloPi