jekhokie/raspberry-noaa-v2 #889: Add support for selection from multiple serial numbered SDR devices not just device_ID

drewid74, 2023-12-20:
**Please Describe the problem to be solved**
When utilizing multiple SDR devices on a pi of the same type, the device ID for each SDR changes if unplugged/plugged in and it seems to change after a reboot as well.  I have been able to update serial number on rtlsdr to differentiate between multiple devices.  In this particular case, another software sometimes grabs the wrong adapter blocking downloads when the cron kicks in since the device is busy.  

**(Optional): Suggest A Solution**
I believe rtlsdr has the option of specifying the device not just by device ID but also by serial. If we can select by serial (if needed on our end)  we could provide the updated serial of the device that we need to use each time with NOAA or Meteor captures.  
* one or other option in settings.yml config file - remark out one you don't need, select or keep standard device ID of 0, or manually update with appropriate serial
* may open up confusion for support if both are enabled - unsure which would likely default to device 0 or serial specified, but could cause no devices to be used
* edge use case where I personally use RaspiNOAA V2 image and configured GOES recv captures.  If I manually start capture with one device unplugged, GOES capture (which has no device selection option) grabs only available one, I can thenk update settings.yml to utilize device ID 1 as the correct one, but this does not allow for remote or scripted system restarts without hands on the box pulling and replacing SDRs. 
* "Get another pi specifically for GOES" - prices are still high and supply is limited, also RaspiNOAA cpu cycles are not utilized in most cases except for during NOAA/Meteor captures. Would be nice to maximize hardware and power consumption to do more.

jp112sdl, 2023-12-23:
> I believe rtlsdr has the option of specifying the device not just by device ID but also by serial

`rtl_fm -help` shows no parameter, that allows to specify serial number. Only a device index (`-d`):
```
rtl_fm, a simple narrow band FM demodulator for RTL2832 based DVB-T receivers

Use:	rtl_fm -f freq [-options] [filename]
	-f frequency_to_tune_to [Hz]
	    use multiple -f for scanning (requires squelch)
	    ranges supported, -f 118M:137M:25k
	[-M modulation (default: fm)]
	    fm, wbfm, raw, am, usb, lsb
	    wbfm == -M fm -s 170k -o 4 -A fast -r 32k -l 0 -E deemp
	    raw mode outputs 2x16 bit IQ pairs
	[-s sample_rate (default: 24k)]
	[-d device_index (default: 0)]
	[-T enable bias-T on GPIO PIN 0 (works for rtl-sdr.com v3 dongles)]
	[-g tuner_gain (default: automatic)]
	[-l squelch_level (default: 0/off)]
	[-p ppm_error (default: 0)]
	[-E enable_option (default: none)]
	    use multiple -E to enable multiple options
	    edge:    enable lower edge tuning
	    dc:      enable dc blocking filter
	    deemp:   enable de-emphasis filter
	    direct:  enable direct sampling 1 (usually I)
	    direct2: enable direct sampling 2 (usually Q)
	    offset:  enable offset tuning
	filename ('-' means stdout)
	    omitting the filename also uses stdout

Experimental options:
	[-r resample_rate (default: none / same as -s)]
	[-t squelch_delay (default: 10)]
	    +values will mute/scan, -values will exit
	[-F fir_size (default: off)]
	    enables low-leakage downsample filter
	    size can be 0 or 9.  0 has bad roll off
	[-A std/fast/lut choose atan math (default: std)]

Produces signed 16 bit ints, use Sox or aplay to hear them.
	rtl_fm ... | play -t raw -r 24k -es -b 16 -c 1 -V1 -
	           | aplay -r 24k -f S16_LE -t raw -c 1
	  -M wbfm  | play -r 32k ... 
	  -s 22050 | multimon -t raw /dev/stdin
```

mbruski, 2024-01-25:
You might want to try this program.

https://github.com/gdrapp/rtl_index_by_serial

It could be run manually after replugging the device or automagically during boot to map serial numbers to device indexes.   The config file would need to have a new variable setup to specify the serial # of the device you want to use for each of the birds (maybe use -1 to indicate when you don't care).   The scripts would also need a modification to check this variable prior to beginning reception and perform the index lookup if and only if a device_serial is assigned to a given bird.  There are probably other ways to use it as well.