jekhokie/raspberry-noaa-v2 #890: [Error]

Tomrm96, 2023-12-23:
**Describe the bug**
A clear and concise description of what the bug is.

Followed the instructions, Raspberry PI 5 but get the following error below. 
TASK [common : install dependencies] ***********************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "No package matching 'libopenblas-base' is available"}

PLAY RECAP *************************************************************************************************************
localhost                  : ok=1    changed=0    unreachable=0    failed=1    skipped=0    rescued=0    ignored=0
**Screenshots and/or logs**

Unsure on how to continue, I have tried to open the web but it doesn't load. 
If applicable, add screenshots to help explain your problem. Logs are always welcome as well.

[image]

**Support Output**
Paste the output of running the `./support.sh` script here.

**Server information (please complete the following information):**
 - OS: [e.g. Raspbian] Latest OS lite 
 - Python version [e.g. 3.7.3]
 - SDR device [e.g. RTL-SDR v3] RTLSDR 4 (Haven'tplugged it in yet)

**Additional context**
Add any other context about the problem here.

Tomrm96, 2023-12-27:
Followed from another ticket some advise given by Jerome, I now get the following error:

[image]

ecriscente, 2024-01-24:
same issue here. any progress solving that?

Tomrm96, 2024-01-24:
Use Satdump, they have automation now. Devs are more responsive and its actively being worked on

closed 2024-01-26 by Tomrm96