jekhokie/raspberry-noaa-v2 #891: Add Support for PHP 8+ and RPM Packages

bradjtrammell, 2024-01-01:
I would like to see this suite become more OS agnostic. This is far in a way a complaint from the way this suite works. I'm just thinking of the bigger picture here.

Everyone has their flavor of choice, and I have found myself on more than one occasion, being the type to prefer Rocky Linux over Ubuntu, having thought, "boy it'd be nice to run this on Rocky". So I'd love to see that be a thing considered in the future. I realize that takes the "raspberry" out of the title, however modifications have already been made to make standard Ubuntu installations possible. So the next logical step would be to "include all". 

Additionally, I personally have noticed that it's dang near impossible to run this suite on PHP 8 (seeing that 7.4 is now EOL). So making it PHP agnostic would also be nice to have, since Ubuntu 22 defaults to 8+, and Ubuntu 20 defaults to 7.4.

The suite has grown by leaps and bounds just in the 6 or so months I've been using it, but I think the ability to choose web server, php version and OS would sky rocket this from niche to THE SOLUTION for NOAA APT.

ImDroided, 2024-05-28:
Yea getting this updated for the newer operating systems would be awesome. actually installing this on my pi5 would be great.

MihajloPi, 2024-12-02:
Hello, sorry for the wait. We'll consider building rpm packages too. Currently it's possible to run RN2 in a virtual environment. I myself run RN2 in Proxmox. I'll see what @creas002 thinks of this.