jekhokie/raspberry-noaa-v2 #893: [Error]

philoswuzgud, 2024-01-05:
**Describe the bug**
webserver: ensure nginx is running

have problem issue during install

TASK [webserver : ensure nginx is running] **************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Unable to start service nginx: Job for nginx.service failed because the control process exited with error code.\nSee \"systemctl status nginx.service\" and \"journalctl -xe\" for details.\n"}

PLAY RECAP **********************************************************************************************************
localhost                  : ok=77   changed=10   unreachable=0    failed=1    skipped=36   rescued=0    ignored=1   

Error:   Something failed with the install - please inspect the logs above

**Screenshots and/or logs**
[image]

**Server information (please complete the following information):**
 - OS: [Raspbian]
 - Python version [e.g. 3.7.3]
 - SDR device [e.g. RTL-SDR v3]

jrjstar, 2024-02-12:
Are you running the 32 bit lite version of Pi OS (terminal only).  If so you should also be seeing an error with any desk earlier in the install.  I would suggest that you install the full version of Pi OS 32-bit (with gui and recommended apps) and install through the terminal as before.  This worked for me, not dude if the lite version is missing some driver that makes anydesk work (this is causing the problem with nginx).  Hope this helps!

jrjstar, 2024-02-13:
Also try using the app image that worked for me