jekhokie/raspberry-noaa-v2 #910: Map overlay problem

yvesclousier, 2024-01-21:
Hello everyone. Great job with this app. 
Since the last version I noticed that the base map does not correspond perfectly with the images received. do you have an idea ? 
See the images here:

http://15.188.192.50/captures?page_no=1

Looking online, I realized that I wasn't the only one with the same problem. thanking you in advance for your return.

Sasszem, 2024-02-24:
Sadly I could not open the link. Is the problem a small offset between the overlaid map and the actuall terrain?

See this picture:
[image]
(The offset is quite noticable at the island above Berlin)

yvesclousier, 2024-02-25:
Hi, 
Sorry for the link, you can check here [https://clousier.com/captures/listImages?pass_id=424]. 
All Noaa captures processed by WxtoImg seem to have this offset.
On the other hand, the reception of Meteor with processing by Satdump does not show this offset.
The server is synchronized well on an NTP server.
[image]

Sasszem, 2024-02-25:
In our case, the problem seems to only happen with Meteor, not with NOAA. Even there, it's inconsistent - we just switched on Meteor on friday, but the images on sunday are lined up correctly. 

I noticed that you also mentioned using RTL-TCP for reception, I wonder if that makes any difference - it should not imho.

MihajloPi, 2024-04-12:
Thanks for reporting the issue! This should have been solved by now so please try updating your OS and the RN2 and run the install script again. In settings.yml file now there is a variable to tune map offsets for NOAA which only happen when decoding with WXtoImg and never with SatDump. The reason it doesn't happen with Meteor is, the satellite transmits its on-board time along with the image. NOAA doesn't so it can suffer from clock on the Pi being inaccurate by a few seconds.

athisha1, 2024-08-15:
> Hello everyone. Great job with this app. Since the last version I noticed that the base map does not correspond perfectly with the images received. do you have an idea ? See the images here: [...]