jekhokie/raspberry-noaa-v2 #911: Issue with accessing database after installing [Error]

chr0nik, 2024-01-21:
Hi,

After successfully install raspberry-noaa-v2 on a Orange PC mini computer running Armbian, I get the following issue:

Fatal error: Uncaught Exception: Unable to open database: unable to open database file in /var/www/wx-new/Lib/Controller.php:46 Stack trace: #0 /var/www/wx-new/Lib/Controller.php(46): SQLite3->__construct() #1 /var/www/wx-new/Lib/Controller.php(12): Lib\Controller->connectToDB() #2 /var/www/wx-new/Lib/Router.php(50): Lib\Controller->__construct() #3 /var/www/wx-new/Lib/Router.php(13): Lib\Router->handleRequest() #4 /var/www/wx-new/public/index.php(16): Lib\Router->__construct() #5 {main} thrown in /var/www/wx-new/Lib/Controller.php on line 46

Can anybody please assist, thanks!

MihajloPi, 2024-04-12:
Please try updating your OS and upsate RN2 to the latest version and run the install script to see if it solves the issue.

jtadesse977, 2024-04-21:
Getting same issue, have been digging around var/www/wx-new/Lib with no luck. Also did full update as well

jtadesse977, 2024-04-23:
YO I FIGURE IT OUT

jtadesse977, 2024-04-23:
Your /home/user directory doesn't have the write/read permissions. If you want to just say screw it you can chmod 777 the home and user directories that way your /var/www directory has access. once you do that it should work but if not lmk

closed 2024-12-02 by MihajloPi