jekhokie/raspberry-noaa-v2 #915: Globe not accurate

mattfox27, 2024-02-03:
Globe in homepage is not showing accurate sats, NOAA's are right overhead and globe shows them on the other side of the world, is there an update or something that needs to be done?

OK1SLM, 2024-02-14:
you need to change the link in the index.php file
raspberry-noaa-v2/webpanel/App/Views/Passes/index.php
https://satvis.space/next/
add "next/" it's there twice
then run ./install_and_upgrade.sh

baz069, 2024-12-06:
To add to this, the tag for the Sensor Cone has also changed slightly.

Original code:
`https://satvis.space/?elements=Point,Label,SensorCone&layers=ArcGis&terrain=None& ... etc`

Working code:
`https://satvis.space/**next/**?elements=Point,Label,**Sensor-cone**&layers=ArcGis&terrain=None& ... etc`