jekhokie/raspberry-noaa-v2 #917: No Passes Scheduled[Error]

jrjstar, 2024-02-11:
**Describe the bug**
Under the time and date section, no passes appear to be scheduled, however when I click on a satellite I can see when it is scheduled to pass over, but this does not show in the scheduled passes section.

**Screenshots and/or logs**
[image]
[image]

**Support Output**
=============================================
Details about environment
=============================================
Current local date/time: Sun 11 Feb 12:42:42 EST 2024
Current date/time (ms):  1707673362
Repo git hash:           b4c6be1e1c53892a8f05186102be6236e6b396e0
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.2
Architecture:            Linux pi-weather 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux
Num CPUs:                4
**---------------------------------------------
'at' Scheduled Jobs (Captures):
  (None)
---------------------------------------------**
Satellite SDR Settings:
  meteor_m2_memory_threshold: 10
  noaa_15_schedule: true
  noaa_15_sdr_device_id: 0
  noaa_15_freq_offset: 0
  noaa_15_enable_bias_tee: false
  noaa_15_gain: 29.7
  noaa_15_sun_min_elevation: 6
  noaa_15_sat_min_elevation: 30
  noaa_18_schedule: true
  noaa_18_sdr_device_id: 0
  noaa_18_freq_offset: 0
  noaa_18_enable_bias_tee: false
  noaa_18_gain: 29.7
  noaa_18_sun_min_elevation: 6
  noaa_18_sat_min_elevation: 30
  noaa_19_schedule: true
  noaa_19_sdr_device_id: 0
  noaa_19_freq_offset: 0
  noaa_19_enable_bias_tee: false
  noaa_19_gain: 29.7
  noaa_19_sun_min_elevation: 6
  noaa_19_sat_min_elevation: 30
  meteor_m2_3_schedule: true
  meteor_m2_3_sdr_device_id: 0
  meteor_m2_3_freq_offset: 0
  meteor_m2_3_enable_bias_tee: false
  meteor_m2_3_gain: 29.7
  meteor_m2_3_schedule_sun_min_elevation: -90
  meteor_m2_3_sun_min_elevation: 6
  meteor_m2_3_sat_min_elevation: 30
  meteor_m2_3_80k_interleaving: false
---------------------------------------------
USB Device Map:
  Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
  Bus 001 Device 003: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
  Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
  Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
---------------------------------------------
Disk Info:
  NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
  mmcblk0     179:0    0 29.8G  0 disk
  ├─mmcblk0p1 179:1    0  2.2G  0 part
  ├─mmcblk0p2 179:2    0    1K  0 part
  ├─mmcblk0p5 179:5    0   32M  0 part
  ├─mmcblk0p6 179:6    0  256M  0 part /boot
  └─mmcblk0p7 179:7    0 27.3G  0 part /
---------------------------------------------
Disk Usage Info:
  Filesystem     1K-blocks    Used Available Use% Mounted on
  /dev/root       28074148 7402256  19222728  28% /
  devtmpfs         1800660       0   1800660   0% /dev
  tmpfs            1965524       0   1965524   0% /dev/shm
  tmpfs             786212     856    785356   1% /run
  tmpfs               5120       4      5116   1% /run/lock
  tmpfs            1024000       0   1024000   0% /var/ramfs
  /dev/mmcblk0p6    258094   50413    207682  20% /boot
  tmpfs             393104       0    393104   0% /run/user/1000
---------------------------------------------
Memory Info:
  MemTotal:        3931052 kB
  MemFree:         3661408 kB
  MemAvailable:    3722468 kB
---------------------------------------------
Git source files changed:
  config/settings.yml
---------------------------------------------
SDR Information:
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:              0x0bda
Product ID:             0x2838
Manufacturer:           Realtek
Product:                RTL2838UHIDIR
Serial number:          00000001
Serial number enabled:  yes
Bias Tee always on:     no
Remote wakeup enabled:  no
__________________________________________
---------------------------------------------
Database tables:
  predict_passes =>
      CREATE TABLE predict_passes(
    sat_name text not null,
    pass_start timestamp primary key default (strftime('%s', 'now')) not null,
    pass_end timestamp default (strftime('%s', 'now')) not null,
    max_elev int not null,
    is_active boolean, pass_start_azimuth int, direction text, azimuth_at_max int, at_job_id int not null default 0);

  decoded_passes =>
      CREATE TABLE decoded_passes(
    id integer primary key autoincrement,
    pass_start integer,
    file_path text not null,
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));

**Server information (please complete the following information):**
 - OS: Pi OS 32-bit Lite
 - Python version: 3.9.2
 - SDR device: RTL-SDR v3
**Additional context**
this is a fresh install of raspberry noaa

jrjstar, 2024-02-12:
It's also worth noting that I tried again with a build of Raspberry Pi OS 32bit (full), and I'm still encountering this error.  I have tried running the schedule command but that didn't work.

ve3elb, 2024-02-12:
Try installing the image. And go from there. As for AnyDesk it may have been upgraded so try installing it manually

(https://qsl.net/ve3elb/RaspiNOAA)

jrjstar, 2024-02-12:
> Try installing the image. And go from there. As for AnyDesk it may have been upgraded so try installing it manually
> 
> (https://qsl.net/ve3elb/RaspiNOAA)

Will try this later, thanks for the help!

jrjstar, 2024-02-13:
This worked!  Thanks again!

closed 2024-02-13 by jrjstar