jekhokie/raspberry-noaa-v2 #920: RTL-TCP reception

Sasszem, 2024-02-23:
**RTL-TCP reception**
It is often better to use an SDR receiver in a remote location, and is often accessed via RTL-TCP.
SATDUMP already has support for it, but the reception scripts do not support it yet.

yvesclousier, 2024-02-25:
Hi, 
I work like this.
I have my PI running with RTL_TCP and I installed RN-V2 on a VPS server at Amazon (AWS).
You must modify the 2 scripts receive_noaa.sh and receive_meteor.sh where Satdump is called for start capture  and remplace the parameter '--source $receiver' by '--source rtltcp --ip_adresse w.x.y.z' where w.x.y.z is the IP adress of your SDR.

It works wonderfully, but I know this modification is not very clean. This receiver type option should be added to the RN-V2 settings. Like receiver_type: 'rtltcp', but I'm not good enough in programming to make all these modifications.
Maybe someone in the community could do it.

Sasszem, 2024-02-25:
We hacked it into the receiver selection switch-case, but that's also not the best solution. I opened the issue so hopefully a proper way will be provided eventually.