jekhokie/raspberry-noaa-v2 #932: [Error] error during installation. wrong file name references in the script

zqushair, 2024-03-03:
**Describe the bug**
When running ./install_and_upgrade.sh on a fresh installation, I kept encountering an error for installing wxtoimg

The problem is that the script refers to a file name that does not exist. The script looks for 
/home/pi/raspberry-noaa-v2/software/wxtoimg-arm64-2.11.2-beta.deb

when the only two files in that directory are amd64 and armhf.  There is no arm64.  I later found that the same bug exists for a couple more packages (satdump and  that referred to the arm64 while the only files available are amd64 and armhf. 

**Screenshots and/or logs**
TASK [common : Installing WXtoImg]  
*********************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "Unable to install package: E:Could not open file /home/pi/raspberry-noaa-v2/software/wxtoimg-arm64-2.11.2-beta.deb - open (2: No such file or directory)"}

PLAY RECAP 
*********************************************************************************************
localhost                  : ok=3    changed=0    unreachable=0    failed=1    skipped=2    rescued=0    ignored=0

Error:   Something failed with the install - please inspect the logs above

**Support Output**
not relevant

**Server information (please complete the following information):**
 - OS: Raspbian bullseye 64bit full
 - Python version 
 - SDR device RTL-SDR V3 and RSP1B

**Solution**
I made a copy of the armhf files and called them arm64. Kept the rest of the file names the same. Ran the install script again and all went well.

zqushair, 2024-03-04:
So I just found out that on a raspberry pi 4 when you run "dpkg --print-architecture", the response is arm64, not armhf. That's why the script ends up looking for a file that does not exist.

AStoker, 2024-08-06:
Any fix for this? Looks like it just needs to have the right version of satdump included in the release?