jekhokie/raspberry-noaa-v2 #934: Animations not working

Lopastudio, 2024-03-07:
Hi. I would like to ask about the animations. I enabled them, and they do not work.
After a little examination, it seems that the application does not even create videos (animations). Is it something I am missing?
I would really like for it to make animations from diffrent passes, so I can see how wind blows etc. 

Thanks

OK1SLM, 2024-03-11:
The noaa_avi.sh script refers to folders and files that I don't think are used anymore

Lopastudio, 2024-03-12:
Is there any way I can use `noaa_avi.sh`  or reenable it?

OK1SLM, 2024-03-14:
The script itself must be modified

Lopastudio, 2024-03-17:
OK1SLM: Ahoj, ako presnejsie "modifikovany", mohol by si sa na to prosim pozrieť, ak zvýši čas? :)

English friends:
Hi, do you mean "modified", could you please look at it if you have the time for that? Thanks :)

PhilRepisky, 2024-04-24:
check/set 3 things in the configuration:

`config/settings.yml`

-->  noaa_decoder: 'wxtoimg' 
 #        (note: default value seen includes list of ALL supported image processors excluding 'avi' which must be explicitly added when      opting to 'enable_animation' below )
--> noaa_daytime_enhancements: 'ZA MCIR MCIR-precip MSA MSA-precip HVCT HVCT-precip NO sea therm avi'
--> noaa_nighttime_enhancements: 'ZA MCIR MCIR-precip NO sea therm avi'

fix a broken IF statement:

`scripts/recieve_noaa.sh`
`scripts/receive_meteor.sh`
(there is a space missing for the brackets)
--> see line 256 https://github.com/jekhokie/raspberry-noaa-v2/blob/master/scripts/receive_noaa.sh

 change it to:   if [ $enhancement == "avi" ]; then

personal preference:
update the avi generated to use the MCIR-precip instead of MCIR

`scripts/image_processors/noaa_avi.sh`
 # produce the output image
 > $WXTOIMG -o -M 49 -m "${MAP_OVERLAY}" ${extra_args} -e "MCIR-precip" "${INPUT_WAV}" "${OUTPUT_IMAGE}"

OK1SLM, 2024-06-22:
Great job, but it would still like the video to be automatically deleted after 24 hours

ShianMull, 2024-07-30:
Hi, Just wondering if anyone can point me to where the animation should be called from?  I can see RollingAnnimation.avi created in the /tmp folder, but browser Passes Page does not seem to link to the created video.  Just need a pointer to tie them together.  Many thanks.  Ps I’m running the Bookworm 64-bit beta version of RaspiNOAA.

OK1SLM, 2024-08-11:
I have the 32 bit version
/srv/videos

ShianMull, 2024-08-11:
Ah! Thanks, but I can see the videos as created. It’s that the web front end does not seem to be pointing to the latest video when video enabled.  I need to know where the website code is located to check where it is expecting to see the video file.

MihajloPi, 2024-12-02:
This is a known issue and proper animation implementation is to be expected.