jekhokie/raspberry-noaa-v2 #958: [Error] Captures don't work

pichu314, 2024-04-27:
Hi, 
Captures don't work or if they work, I only have noise. I checked `output.log` and I got this : 
```
rm: cannot remove './dataset.json': No such file or directory
rm: cannot remove './product.cbor': No such file or directory
27-04-2024 10:21 /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Files recorded
27-04-2024 10:21 /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Resampling down audio
/usr/bin/sox WARN wav: Premature EOF on .wav input file
/usr/bin/sox WARN rate: rate clipped 184188 samples; decrease volume?
/usr/bin/sox WARN dither: dither clipped 161166 samples; decrease volume?
27-04-2024 10:21 /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : Deleting audio files
27-04-2024 10:21 /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh INFO : No images found - not pushing anywhere
```

---
```bash
> ./support.sh
=============================================
Details about environment
=============================================
Current local date/time: Sat 27 Apr 10:37:55 CEST 2024
Current date/time (ms):  1714207075
Repo git hash:           79bb36a3d5e89e5017cc754b7d16d9f41100ad8e
Raspberry Pi Model:      Raspberry Pi 4 Model B Rev 1.4
Architecture:            Linux noaa 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Num CPUs:                4
---------------------------------------------
'at' Scheduled Jobs (Captures):
  * [44 Tue Apr 30 10:28:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240430-082855 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714465735 938 44 Southbound W
  * [33 Mon Apr 29 22:53:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20240429-205310 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714423990 947 59 Northbound W
  * [62 Thu May  2 11:22:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20240502-092257 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714641777 975 63 Southbound E
  * [60 Thu May  2 23:35:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20240502-213531 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714685731 969 58 Northbound E
  * [51 Tue Apr 30 15:55:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 4" METEOR-M2-4-20240430-135559 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714485359 968 53 Northbound E
  * [64 Fri May  3 20:33:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240503-183323 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714761203 933 72 Northbound E
  * [20 Sun Apr 28 12:57:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20240428-105727 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714301847 967 69 Southbound E
  * [37 Sun Apr 28 11:15:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20240428-091519 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714295719 932 55 Southbound E
  * [45 Tue Apr 30 20:10:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240430-181055 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714500655 939 46 Northbound E
  * [67 Fri May  3 12:16:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20240503-101637 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714731397 958 49 Southbound W
  * [65 Fri May  3 13:35:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20240503-113546 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714736146 973 57 Southbound W
  * [9  Sun Apr 28 21:02:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240428-190202 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714330922 910 64 Northbound W
  * [66 Fri May  3 23:23:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20240503-212313 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714771393 929 46 Northbound E
  * [59 Thu May  2 13:48:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 18" NOAA-18-20240502-114818 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714650498 932 45 Southbound W
  * [49 Tue Apr 30 22:40:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20240430-204049 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714509649 987 74 Northbound W
  * [41 Sat Apr 27 17:00:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 4" METEOR-M2-4-20240427-150017 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714230017 918 45 Northbound W
  * [57 Thu May  2 20:59:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240502-185903 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714676343 911 67 Northbound W
  * [54 Wed May  1 22:28:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20240501-202831 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714595311 969 86 Northbound E
  * [36 Sat Apr 27 11:38:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 3" METEOR-M2-3-20240427-093821 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714210701 942 86 Southbound E
  * [43 Mon Apr 29 16:17:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_meteor.sh "METEOR-M2 4" METEOR-M2-4-20240429-141713 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714400233 937 79 Northbound E
  * [30 Sun Apr 28 11:37:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20240428-093718 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714297038 960 82 Southbound E
  * [56 Thu May  2 09:37:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240502-073726 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714635446 933 67 Southbound E
  * [68 Fri May  3 22:04:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20240503-200408 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714766648 930 54 Northbound E
  * [61 Thu May  2 22:16:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 19" NOAA-19-20240502-201618 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714680978 950 68 Northbound E
  * [52 Wed May  1 10:03:00 2024 a nugget] -> /home/nugget/raspberry-noaa-v2/scripts/receive_noaa.sh "NOAA 15" NOAA-15-20240501-080308 /home/nugget/raspberry-noaa-v2/tmp/orbit.tle                                                               1714550588 908 73 Southbound W
[... cut ...]
    daylight_pass boolean, is_noaa boolean, sat_type integer, img_count integer, has_spectrogram boolean default 0, has_pristine boolean default 0, gain real, has_polar_az_el boolean default 0, has_polar_direction boolean default 0, has_histogram boolean default 0,
    foreign key(pass_start) references passes(pass_start));
```
---
**Server information**
 - OS: **Raspbian Lite 64bit**
 - Python version **3.9.2**
 - SDR device : **RTL-SDR v4**

I don't understand because it used to be working before...
I precise that I use a V-dipole antenna with a **SWR** of **1.5**, and **impedance** of **43Ω** without **LNA** 
Thanks in advance for your help :)

PhilRepisky, 2024-04-27:
if you run `satdump live` when the satellite is overhead can you produce a waveform? 

note that a V-dipole should be facing the direction of the pass ( |< when sat going -->) and ( >| when sat going <--- )

example of a run from today on my device:
`/usr/bin/satdump live noaa_apt . --source rtlsdr --samplerate 1.024e6 --frequency 137.6200e6 --satellite_number 15 --source_id 0 --gain 31.0  --start_timestamp 1714223568  --sdrpp_noise_reduction  --http_server 0.0.0.0:8081  --timeout 923 >> /var/log/raspberry-noaa-v2/output.log 2>&1`

of course this needs to be adjusted to your own test, these are now numbers in the past.

pichu314, 2024-04-30:
I don't really know how satdump works. But the issue is not here. Because even with **WXtoIMG**, I have nothing, only **noise**. 
Furthermore, the **spectrogram** doesn't show anything. 
I don't know if there is a link but when I use the `test_reception.sh`, I don't heard anything when I'm connect with `ncat`. (I precise that I know how this works cause I had already do this in the past). 
My antenna seems totally correct. I didn't touch it and before it worked. I test it and it has **SWR** of **1.3**, and **impedance** of **43Ω**. I really don't know what I did... :(

pichu314, 2024-04-30:
Also, I checked prediction pass on **AMSAT** and I see the next pass in **UTC** for my location is at **19:51**.
[image]

But, on the **Raspberry**, in **UTC+2**, I see the pass at **20:10** when it should be **21:51:** 
[image]

Knowing that I have this empty **spectrogram**, maybe the capture doesn't start at the right time. 
[image]

PhilRepisky, 2024-05-01:
ya, that would confirm that at a minimum you aren't configured correctly somewhere. possibly `raspi-config` to check localisation options? I would guess you have a 30 minute offset

note that an additional 10 minute difference can also come up based on the minimum sat angle you're assuming, the higher the cutoff the later in the pass you're scheduled for (vs the AMSAT lookup).

if you've mis-assigned your location, double check your lat / long in the `settings.yml` then re-run the `install_update` script.

I wouldn't expect the TLEs to be a problem, but its odd that gpredict would act up. perhaps run that directly in the terminal as well against a NOAA TLE? 

(in `schedule_captures` you can see the CLI usage `  predict_start=$($PREDICT -t $TLE_FILE -p "${OBJ_NAME}" "${next_predict}" | head -1)'.

that aside, if you're having trouble with the timing we can at least confirm your antenna + line-of-sight-to-sky + other implementation details with the prior suggested solution of running satdump directly (or `satdump-ui` if you have a monitor to click around in)

pichu314, 2024-05-01:
Ok, I try with `predict` :  
```bash
user@noaa:~ $ predict -t /home/user/.config/satdump/satdump_tles.txt -p "NOAA 19" | head -1
1714589415 Wed 01May24 18:50:15    0  110  231   31  330   3387  78481 * 0.000000
```
Results correlate with **AMSAT** : 
[image]
But, on the server with **UTC+2** : 
[image]
I checked **lat / long** on `settings.yml `and it's good.
I'm waiting for the next pass to try with **SatDump**.

closed 2024-05-17 by radionugget